sstsimulator/sst-elements #681: Is there any ways to get some performance matrices like CPI(cycle per instruction) from Statistic API when running executable on Ariel ?

yeungsl, 2017-06-14:
New Issue for sst-elements
--------------------------

1 - I want to get some Statistics on some of the performance metrics from running some benchmark on Ariel. e.g. : CPI (cycle per instruction), Memory Stalls Cycles, Cache misses per killo instructions. I found Cache Tracker but that is not what I wanted. I think McNiagara Element Library may does something similar but that is off the repo right now. So Just wondering is there some other components or Stats API that I can use to get those stats I need?

3 - I am using Ubuntu 14.04 with Pin tool 2.14 and SST 7.1.0

nmhamster, 2017-06-16:
Hi @yeungsl - the answer to this is yes, we need these statistics for our own work too. In order to get most of the statistics out from our models you should add the following code to the **end of your Python script/SST model.**

```
# Enable SST Statistics Outputs for this simulation
sst.setStatisticLoadLevel(16)
sst.enableAllStatisticsForAllComponents({"type":"sst.AccumulatorStatistic"})
sst.setStatisticOutput("sst.statOutputCSV")
sst.setStatisticOutputOptions( {
               	"filepath"  : "./stats-ariel.csv",
               	"separator" : ", "
        } )
```

This will turn on basic statistics gathering for all of the components in your simulation (you can individually enable these etc, but for now lets just collect them all).

If you run your simulation, it will output a file: 

Take a look in your CSV output. For Ariel models you'll see something like this (my values are all zero but this is meant to just show you how to get the counters):

```
A0, tlb_hits, , Accumulator, 1073043468, 0, 0, 0, 0
A0, tlb_evicts, , Accumulator, 1073043468, 0, 0, 0, 0
A0, tlb_translate_queries, , Accumulator, 1073043468, 0, 0, 0, 0
A0, tlb_shootdown, , Accumulator, 1073043468, 0, 0, 0, 0
A0, tlb_page_allocs, , Accumulator, 1073043468, 0, 0, 0, 0
A0, read_requests, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, write_requests, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, read_request_sizes, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, write_request_sizes, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, split_read_requests, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, split_write_requests, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, no_ops, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, instruction_count, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_sp_ins, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_dp_ins, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_sp_simd_ins, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_dp_simd_ins, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_sp_scalar_ins, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_dp_scalar_ins, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_sp_ops, 0, Accumulator, 1073043468, 0, 0, 0, 0
A0, fp_dp_ops, 0, Accumulator, 1073043468, 0, 0, 0, 0
```

You should be able to determine the number of instructions you executed here. The cycle count will come from the time the simulation ran to (this is the `1073043468` number in this file). You should be able to use the statistics from the processor models, caches etc to get most of the basic hardware information you want from the simulation.

Please let us know if this gets you what you need or if you have additional questions etc. Thanks for your interest in SST!

yeungsl, 2017-06-16:
Thank you so much. But I wonder is there anyways to collect Stats with respect to instructions CPU ran rather than time? I know in Stats API I can just run 
`StateExample0.enableStatistics(["stat1_U32"], {
      "type" : "sst.HistogramStatistic",
      "rate" : "100 ms"})`
However, "rate" only accepts time or events. So if I were to collect some data from Cache for example cache misses, I can get cache misses/ 100 ms. But rather is there a way I can get cache misses / 100 instruction that CPU ran.

allevin, 2020-04-03:
No further response from author, considering this issue closed.

closed 2020-04-03 by allevin