sstsimulator/sst-elements #682: Miranda (SingleStreamGenerator) Only Issues One Request Per Cycle

hughes-c, 2017-06-16:
1 - Detailed description of problem or enhancement
The SingleStreamGenerator in Miranda only issues one request per cycle.

2 - Describe how to reproduce
```
comp_cpu_0.addParams({
   "verbose" : 0,
   "clock" : core_clock,
   "max_reqs_cycle": 2,				
   "max_reorder_lookups" : 16,
   "maxmemreqpending" : 16,
   "generator" : "miranda.SingleStreamGenerator",
   "generatorParams.verbose" : 1,
   "generatorParams.startat" : 0,
   "generatorParams.count" : 50000000,
   "generatorParams.max_address" : int(maxSize) * 1024,     
   "generatorParams.length" : int(stride),
   "printStats" : 1,
})
```

3 - What Operating system(s) and versions 
RHEL 6

4 - What version of external libraries (Boost, MPI)
Boost - 1.56
OpenMPI 1.8.8

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
sst-elements@https://github.com/sstsimulator/sst-elements/commit/32f2edb9905e341d55e7cf8793ed35c1b765b123
https://github.com/sstsimulator/sst-core/commit/543e0d0f13d6430d247c5cf4854c6d918c4a282f

nmhamster, 2017-06-23:
@hughes-c - can you see if this addresses the issue you had raised? Thanks.

hughes-c, 2017-06-26:
@nmhamster Yes, it appears to have corrected the problem.

nmhamster, 2017-06-26:
@hughes-c - thanks! Closing this issue.

closed 2017-06-26 by nmhamster