sstsimulator/sst-elements #686: Is there any way for SST to run a Matrix benchmark faster?

yeungsl, 2017-06-18:
New Issue for sst-elements
--------------------------

I set up an configuration with Ariel with 8 cores, 64 KB per core l1 cache(memHierachy.Cache), 512 KB l2 cache shared, 16 MB l3 cache and 16 GB main memory (memHierarchy.MemController). I am running an 2D matrices multiplication from [polybenchmark](http://web.cse.ohio-state.edu/~pouchet.2/software/polybench/). But it never returns (it is alive and terminates with outputs when I press ctrl-C), whereas, on my local pc (4 core, 8GB RAM) it just takes around 10 seconds. Is this normal that I have to wait longer or it is just that I did the wrong configuration?

nmhamster, 2017-06-18:
Hi @yeungsl - how long have you left SST for? Its likely that its running and taking a long time. Does SST exit and print that it is completed or is it just running and not sending any output to the screen?

yeungsl, 2017-06-18:
For one whole day. I have lost that output, but here is an similar output that I just ran for 5 min:
`^CEMERGENCY SHUTDOWN (0,0)!
# Simulated time:                  69.687 ms

Ariel Memory Management Statistics:
---------------------------------------------------------------------
Page Table Sizes:
- Map entries         7724
Page Table Coverages:
- Bytes               31637504
EMERGENCY SHUTDOWN Complete (0,0)!
Simulation is complete, simulated time: 0 s
`
FYI -- the only difference for the whole day simulation is the simulated time is around 16 seconds and Page table coverage is larger.

nmhamster, 2017-06-18:
@yeungsl - OK SST hasn't processed any events. Can you check the Ariel parameter called `ariel_enable` what is this set to in your script?

yeungsl, 2017-06-18:
here is the configuration for ariel`ariel = sst.Component("a0", "ariel.ariel")
ariel.addParams({
        "verbose" : "0",
        "maxcorequeue" : "256",
        "maxissuepercycle" : "2",
        "pipetimeout" : "0",
        "corecount" : args.corecount,
        "executable" : "/nethome/syeung8/sst_config/exectuable/2mm_base",
        "arielmode" : "1",
        "clock" : args.clock,
        "memmgr.memorylevels" : "1",
        "memmgr.defaultlevel" : "0"
        })
`

nmhamster, 2017-06-18:
I'm sorry, I mis read your output, SST is processing events actually (the first simulated time number shows us that). Have you tried a very small test case with your script yet?

yeungsl, 2017-06-18:
Yes the stream benchmark works well

nmhamster, 2017-06-18:
OK, I suspect the executable may just be taking a long time because of the complexity. Is there a section of this benchmark that you are interested in timing/performing analysis on? We typically find that there are a lot of startup regions etc in our benchmarks which we don't want to simulate, and that there is a small section of interest. Is that the case for you too?

yeungsl, 2017-06-18:
If so is there any other option other than rewriting the benchmarks?

hughes-c, 2017-06-19:
You don't really need to "rewrite" the benchmark but you will need the source code so that you can recompile it. You can issue a call to ariel_enable() in front of the section where you want to begin profiling. You will also want to change arielmode to "0" or "2" in the configuration.

Adding additional components will likely exacerbate the problem since you will be simulating additional structures.

yeungsl, 2017-06-19:
Sounds great, but can you be more specific on that? is there a library that I can include in source code(written in C) in order to call ariel_enable()?

hughes-c, 2017-06-19:
You can create a dummy function like:
```
volatile void ariel_enable()
{
     printf("Begin profiling \n");
}
```

nmhamster, 2017-06-19:
There are functions and library supplied in the SST source at: `src/sst/elements/ariel/api` for calling the `ariel_enable()` API.

closed 2017-06-22 by yeungsl