sstsimulator/sst-elements #688: Ariel cores not full utilized

yeungsl, 2017-06-21:
New Issue for sst-elements
--------------------------

I ran a couple of executable on the same configuration where Ariel is used as CPU with 8 cores by default and with l1, l2 caches linked through memBus. However, when I print all the Statistics, it seems to me that only one of cores and its corresponding caches has done the work, because the statistics from other cores and caches are all 0; the executable exits correctly and the simulation ends correctly. Is there any apparent mistake that I am making?

nmhamster, 2017-06-21:
HI @yeungsl, are you running a multi-threaded code?

yeungsl, 2017-06-21:
NO I don't think so

nmhamster, 2017-06-22:
@yeungsl - if you aren't running a threaded binary then the other cores will all return zero for their counters because there is no work to be done on them. We detect the presence of threads running in the binary and map these onto the cores for you, if you only have one thread then only one core will do any work (and have any counters above zero). Let us know if you want a more detailed explanation.

yeungsl, 2017-06-22:
Thanks @nmhamster, glad to confirm what I originally thought.

nmhamster, 2017-06-22:
I'll close this issue, please let us know if you have any other questions.

closed 2017-06-22 by nmhamster