sstsimulator/sst-elements #689: Is there a way to get Clock cycle count per instruction type for Ariel?

yeungsl, 2017-06-22:
New Issue for sst-elements
--------------------------

In the Statistic output of Ariel I can only find number of instructions ran by the CPU and some TLB stats. But I would like to have some clock cycle count statistics from the simulation so that I can calculate CPI for the program I ran. Is there any way of printing the clock cycle out?

nmhamster, 2017-06-22:
@yeungsl - the current version of the develop branch for SST has this statistic included. It is not in the 7.1 version of the code.

yeungsl, 2017-06-22:
Its on devel branch right? Thanks I will try to build that one.

nmhamster, 2017-06-22:
@yeungsl - yes. We have not done any extensive evaluation on the counter yet but we added the code because you had indicated you needed it in another post. It should count cycles for whenever Ariel is active.

yeungsl, 2017-06-23:
Thanks, but I cannot compile it since I did not see any Configure (the executable) in the repo. I copied the source into the tar file you guys released online and ran Configure, but it complains there is no Makefile.in, should I copy all the Makefile.in into all the directory also?

yeungsl, 2017-06-23:
Problem solved. I just added the code you guys changed on the branch into the source . Everything works fine. Thank you very much

closed 2017-06-23 by yeungsl