sstsimulator/sst-elements #696: SST memory statistics producing only 0's in output

jjwilke, 2017-07-05:
Working off of sst-core/sst-elements devel branch from end of June:

Compiler and platform is:
````
gcc --version >&5
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
````

sst-core configured with:
````
./configure --prefix=/home/local/UFAD/aravindneela/local/sstcore --with-boost=/home/local/UFAD/aravindneela/local/packages/boost-1.56
````

sst-elements configured with:
````
./configure --prefix=/home/local/UFAD/aravindneela/local/sstelements --with-sst-core=/home/local/UFAD/aravindneela/local/sstcore --with-dramsim= --with-nvdimmsim= --with-hybridsim=/home/local/UFAD/aravindneela/local/packages/HybridSim --with-pin=/home/local/UFAD/aravindneela/local/packages/pin-2.14-71313-gcc.4.4.7-linux
````

Output of Ariel is:
````
Ariel Memory Management Statistics:
---------------------------------------------------------------------
Page Table Sizes:
- Map entries         0
Page Table Coverages:
- Bytes               0
PERFORMING SAVE OF CACHE TABLE!!!
got to save state in nvdimm 
save file was state/nvdimm_restore.txt
NVDIMM is saving the used table, dirty table and address map 
TLB Misses: 0
TLB Hits: 0
Total prefetches: 0
Unused prefetches in cache: 0
Unused prefetch victims: 0
Prefetch hit NOPs: 0
Prefetch cheat count: 0
Unique one misses: 0
Unique stream buffers: 0
Stream buffers hits: 0
PERFORMING SAVE OF CACHE TABLE!!!
got to save state in nvdimm 
save file was state/nvdimm_restore.txt
NVDIMM is saving the used table, dirty table and address map 
TLB Misses: 0
TLB Hits: 0
Total prefetches: 0
Unused prefetches in cache: 0
Unused prefetch victims: 0
Prefetch hit NOPs: 0
Prefetch cheat count: 0
Unique one misses: 0
Unique stream buffers: 0
Stream buffers hits: 0
PERFORMING SAVE OF CACHE TABLE!!!
got to save state in nvdimm 
save file was state/nvdimm_restore.txt
NVDIMM is saving the used table, dirty table and address map 
TLB Misses: 0
TLB Hits: 0
Total prefetches: 0
Unused prefetches in cache: 0
Unused prefetch victims: 0
Prefetch hit NOPs: 0
Prefetch cheat count: 0
Unique one misses: 0
Unique stream buffers: 0
Stream buffers hits: 0
````

[... 267 lines cut ...]
		dc.addParams(dc_params)

		memLink = sst.Link("mem_link_" + str(next_memory_ctrl_id))
		memLink.connect((mem, "direct_link", ring_latency), (dc, "memory", ring_latency))

		netLink = sst.Link("dc_link_" + str(next_memory_ctrl_id))
		netLink.connect((dc, "network", ring_latency), (router_map["rtr." + str(next_network_id)], "port2", ring_latency))

		next_network_id = next_network_id + 1
		next_memory_ctrl_id = next_memory_ctrl_id + 1

# ===============================================================================

# Enable SST Statistics Outputs for this simulation
sst.setStatisticLoadLevel(16)
sst.enableAllStatisticsForAllComponents({"type":"sst.AccumulatorStatistic"})

sst.setStatisticOutput("sst.statOutputCSV")
sst.setStatisticOutputOptions( {
	"filepath"  : "./stats-snb-ariel-dram.csv",
	"separator" : ", "
} )

print "Completed configuring the SST Sandy Bridge model"
````

gvoskuilen, 2017-07-05:
Can you post the Ariel output before the "Ariel Memory Management Statistics:"? Is there an 'ariel_enable()' call in the application you are running?

nmhamster, 2017-07-05:
@gvoskuilen - the deck is assuming `ariel_enable()` is called so that would be my guess too.

aravindneela, 2017-07-17:
Hi, 
@nmhamster  - The ariel_enable in the python script is set to 0 which means it is enabled. Please correct me if I am wrong.

@gvoskuilen  - The output before Ariel management is:

Configuring Ring Network-on-Chip...
Configuring core and memory controller group 0...
Creating L3 cache block 0...
Creating Core 0 in Group 0
Creating L3 cache block 1...
Creating Core 1 in Group 0
Configuring core and memory controller group 1...
Creating L3 cache block 2...
Creating Core 0 in Group 1
Creating L3 cache block 3...
Creating Core 1 in Group 1
Configuring core and memory controller group 2...
Creating L3 cache block 4...
Creating Core 0 in Group 2
Creating L3 cache block 5...
Creating Core 1 in Group 2
Configuring core and memory controller group 3...
Creating L3 cache block 6...
Creating Core 0 in Group 3
Creating L3 cache block 7...
Creating Core 1 in Group 3
Completed configuring the SST Sandy Bridge model
ArielComponent[arielcpu.cc:52:ArielCPU] Creating Ariel component...
ArielComponent[arielcpu.cc:55:ArielCPU] Configuring for 8 cores...
ArielComponent[arielcpu.cc:58:ArielCPU] Configuring for check addresses = no
ArielComponent[arielcpu.cc:140:ArielCPU] Loading memory manger: ariel.MemoryManagerSimple
ArielComponent[arielcpu.cc:148:ArielCPU] Memory manager construction is completed.
ArielComponent[arielcpu.cc:179:ArielCPU] Model specifies that there are 0 application arguments
ArielComponent[arielcpu.cc:186:ArielCPU] Interception and re-instrumentation of multi-level memory calls is DISABLED.
ArielComponent[arielcpu.cc:194:ArielCPU] Tracking the stack and dumping on malloc calls is DISABLED.
ArielComponent[arielcpu.cc:199:ArielCPU] Base pipe name: /sst_shmem_24476-0-1681692777
ArielComponent[arielcpu.cc:210:ArielCPU] Processing application arguments...
ArielComponent[arielcpu.cc:313:ArielCPU] Completed processing application arguments.
ArielComponent[arielcpu.cc:320:ArielCPU] Creating core to cache links...
ArielComponent[arielcpu.cc:330:ArielCPU] Creating processor cores and cache links...
ArielComponent[arielcpu.cc:333:ArielCPU] Configuring cores and cache links...
ArielComponent[arielcpu.cc:357:ArielCPU] Registering ArielCPU clock at 2660MHz
ArielComponent[arielcpu.cc:361:ArielCPU] Clocks registered.
ArielComponent[arielcpu.cc:369:ArielCPU] Completed initialization of the Ariel CPU.
l3cache_0: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 3 cycles.
l2cache_0: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 2 cycles.
l3cache_1: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 3 cycles.
l2cache_1: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 2 cycles.
Creating DRAM with /home/local/UFAD/aravindneela/scratch/src/DRAMSim2/ini/DDR3_micron_32M_8B_x4_sg125.ini
== Loading device model file '/home/local/UFAD/aravindneela/scratch/src/DRAMSim2/ini/DDR3_micron_32M_8B_x4_sg125.ini' ==
== Loading system model file '/home/local/UFAD/aravindneela/scratch/src/sst-elements/src/sst/elements/memHierarchy/tests/system.ini' ==
WARNING: UNKNOWN KEY 'DEBUG_TRANS_FLOW' IN INI FILE
===== MemorySystem 0 =====
CH. 0 TOTAL_STORAGE : 2048MB | 1 Ranks | 16 Devices per rank
===== MemorySystem 1 =====
CH. 1 TOTAL_STORAGE : 2048MB | 1 Ranks | 16 Devices per rank
Creating Flash with /home/local/UFAD/aravindneela/scratch/src/NVDIMMSim/src/ini/samsung_K9XXG08UXM_gc_test.ini
Done with creating memories
l3cache_2: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 3 cycles.
l2cache_2: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 2 cycles.
l3cache_3: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 3 cycles.
l2cache_3: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 2 cycles.
Creating DRAM with /home/local/UFAD/aravindneela/scratch/src/DRAMSim2/ini/DDR3_micron_32M_8B_x4_sg125.ini
== Loading device model file '/home/local/UFAD/aravindneela/scratch/src/DRAMSim2/ini/DDR3_micron_32M_8B_x4_sg125.ini' ==
== Loading system model file '/home/local/UFAD/aravindneela/scratch/src/sst-elements/src/sst/elements/memHierarchy/tests/system.ini' ==
WARNING: UNKNOWN KEY 'DEBUG_TRANS_FLOW' IN INI FILE
===== MemorySystem 0 =====
CH. 0 TOTAL_STORAGE : 2048MB | 1 Ranks | 16 Devices per rank
===== MemorySystem 1 =====
CH. 1 TOTAL_STORAGE : 2048MB | 1 Ranks | 16 Devices per rank
[... 94 lines cut ...]
Total prefetches: 0
Unused prefetches in cache: 0
Unused prefetch victims: 0
Prefetch hit NOPs: 0
Prefetch cheat count: 0
Unique one misses: 0
Unique stream buffers: 0
Stream buffers hits: 0
PERFORMING SAVE OF CACHE TABLE!!!
got to save state in nvdimm
save file was state/nvdimm_restore.txt
NVDIMM is saving the used table, dirty table and address map
TLB Misses: 0
TLB Hits: 0
Total prefetches: 0
Unused prefetches in cache: 0
Unused prefetch victims: 0
Prefetch hit NOPs: 0
Prefetch cheat count: 0
Unique one misses: 0
Unique stream buffers: 0
Stream buffers hits: 0
Simulation is complete, simulated time: 208.929 ms

I have replaced the actual output from the program with "*program output*" since it was a very long output. The application was executed perfectly but the memory statistics is still giving trouble.

gvoskuilen, 2017-07-24:
@aravindneela The above shows that ariel is configured to wait until it finds an 'ariel_enable()' call in the program - "ARIEL: Tool is configured to suspend profiling until program control". Ariel is not finding this call in your program so Ariel is never triggered to start simulating. There are two ways to fix this - set 'arielmode' to '1' in the config, or add a call to 'ariel_enable()' in your program. The ariel API, including 'ariel_enable()' can be found in the sst-tools repo in sst-tools/tools/ariel/api/ariel.h (use the devel branch from github).

aravindneela, 2017-07-27:
Thanks @gvoskuilen. That worked! 
I have another doubt. I am pasting part of the output from the terminal after running the script (snb-ariel-dram.py) using SST below (in italics):
_PERFORMING SAVE OF CACHE TABLE!!!
TLB Misses: 0
TLB Hits: 0
Total prefetches: 0
Unused prefetches in cache: 0
Unused prefetch victims: 0
Prefetch hit NOPs: 0
Prefetch cheat count: 0
Unique one misses: 0
Unique stream buffers: 0
Stream buffers hits: 0_

This set repeats 4 times and is always 0. Is this normal? Is there any documentation I can refer to understand these outputs?

gvoskuilen, 2017-07-28:
@aravindneela That output is from HybridSim so I would take a look at their documentation. Four repetitions is probably one from each of the HybridSim instances.

taniabanerjee, 2017-08-02:
@gvoskuilen Another question on this same setup. How does one execute mpirun with Ariel? An earlier version of SST (perhaps 6.0) did allow it to be specified using the "executable" parameter to Ariel (e.g. "executable": "mpirun -n 4 /home/tania/nek5000"), but that does not work with the latest devel version.

taniabanerjee, 2017-08-08:
@gvoskuilen Our application CMT-bone is MPI based. So we are looking for ways to simulate MPI programs.  In your paper "The Potentials and Perils of Multi-Level Memory", you used number of parallel codes, though, such as miniFE, LULESH, MineAero, and RSBench. Did you have OpenMP versions for these?

gvoskuilen, 2017-08-08:
@taniabanerjee Yes we were running OpenMP versions of those.

taniabanerjee, 2017-08-10:
@gvoskuilen Is there any other way to execute pure MPI programs (say 4 ranks per node) through SST configured for multi-level memories?

aravindneela, 2017-08-31:
Hi @gvoskuilen,
I have a doubt with respect to running the snb-ariel-dram.py script on sst. While running the command "sst -n 4 snb-ariel-dram.py", is the -n used for running sst in parallel with 4 cores or is it used for simulating the application on 4 processor of the system being simulated?

gvoskuilen, 2017-09-07:
@taniabanerjee No there is not unfortunately. If you had traces of each MPI rank you could run them using Prospero, but it's a simple memory-trace execution model and I'm not sure if it would be sufficient for what you want. Also, Prospero does not have any support for multi-level memory (e.g., the ability to map pages and allocations to different memories) so you may need to add that if you went that route.

@aravindneela The 'n' is used for running SST in parallel with 4 threads.

allevin, 2020-10-14:
This is fixed.

closed 2023-10-24 by hughes-c