sstsimulator/sst-elements #698: How is the L1 cache simulated, why it is different from other caches like L2, L3 etc.?

yeungsl, 2017-07-13:
New Issue for sst-elements
--------------------------
Recently I am running some benchmarks on SST with the configuration of 1 core Ariel, L1 cache, L2 cache, L3 cache and memory. I ran the same executable on both SST and local machine with Linux Perf. But I get very different Cache-miss ratio and CPI. I understands that CPI may vary depends on the processors I use. But I am confused about this variation of Cache-miss ratio. For example, the local machines show that 45% of all requests went out of Cache. But, on SST, more than half of the requests sent to Ariel went to L1 Cache and in the end, only 10% of the request went to the memory controller. I tried multiple benchmarks with different data set size on different configuration of cache sizes on SST as well. But I am observing this every time. 

Furthermore, I took out L2 and L3. Then run  those executable with only 1KB of L1 cache and 4 GB of memory. I still see more than half of the requests went to L1 Cache. So is my understanding of the SST stat "L1.CacheHits", "L1.CachMisses" wrong? Or its just that L1 is implemented with some smart mechanism that always pre-fetches pages? 

FYI: I am using the default output of SST stat, which has Sum, SumSQ, counts all the same on those CacheHits, and CacheMisses output.

amroawad2, 2017-07-13:
@yeungsl :  I believe the difference could result form counting partial misses as misses. Specifically, if an access to a cache line causes a miss and the very next instruction accesses the same cache line, the second access is counted as a hit in SST even though the actual cache line hasn't arrived yet. This might be not the case in real system where partial misses are counted as misses. 

Here is the code for how SST handles the first miss and allocates the block (if I understand it correctly) :
https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/cacheController.cc#L92

This is just my expectation and I would suggest to wait until @nmhamster or @gvoskuilen confirm.

yeungsl, 2017-07-14:
Thanks for those explanations. It's very helpful :)

gvoskuilen, 2017-07-19:
@yeungsl : In the case of partial misses, there is a breakdown of the hits/misses in the statistics - look for Hit_Arrival, Hit_Blocked, Miss_Arrival, Miss_Blocked. Hit_Arrival and Miss_Arrival are hits/misses which did not need to block for another access to the same line (hit/miss on arrival). Hit_Blocked and Miss_Blocked are requests which blocked for an outstanding access and were hits and misses, respectively, when retried. The CacheMisses statistic only counts accesses that generated a miss (Miss_Blocked and Miss_Arrival), CacheHits counts the others.

nmhamster, 2017-11-20:
@yeungsl - closing this issue, please reopen if you have more questions that the team can help with.

closed 2017-11-20 by nmhamster