sstsimulator/sst-elements #700: MemHierarchy should not segfault on misaligned loads

bjmoor, 2017-07-16:
`L1CoherenceController` can segfault when the cache receives a request that spans across cache lines.  It would be nicer to Output.fatal() with such a message, rather than segfault.

nmhamster, 2017-07-17:
@bjmoor @gvoskuilen - I think this depends on the contract with the processor core people. If the *requirement* is that the core ensures no unaligned/split requests then its OK to not have this checked. It might also be better to not have the checks slowing down the behavior? Thoughts?

gvoskuilen, 2017-07-17:
@nmhamster @bjmoor - I think it should be a requirement for the core to ensure no unaligned/split requests. I'd rather avoid an extra check on every access.

bjmoor, 2017-07-18:
Personally, I think that branch predictors are pretty good, and segfaults are bad.  Users should not get a segfault using our code.  That puts a significant onus on them how to debug code that isn't theirs, and when I was receiving that segfault, it was FAR from obvious that the reason was mis-aligned cache accesses.

allevin, 2020-10-14:
Fixed if user turns on debug.

closed 2020-10-14 by allevin