sstsimulator/sst-elements #701: Ariel core cycle counter

yeungsl, 2017-07-17:
New Issue for sst-elements
--------------------------
What cycle did Ariel cycle counter really counts? Did it only includes the cycles where CPU processes events and not counting the Memory Stall cycles where it waits for pages from Memory? Can I make the counter also count the memory stall cycles by adding currentCycles after this line [https://github.com/sstsimulator/sst-elements/blob/ariel-add-cycle-counter/src/sst/elements/ariel/arielcore.cc#L668](https://github.com/sstsimulator/sst-elements/blob/ariel-add-cycle-counter/src/sst/elements/ariel/arielcore.cc#L668 )?

amroawad2, 2017-07-17:
Hi @yeungsl , 
The memory stall cycles will eventually impact the number of cycles spent on CPU. If a memory instruction misses, then it takes longer time waiting, thus the overall number of cycles in Ariel will reflect all other stalls. In other words, if you simulate 100M instructions, the overall CPU cycles will be higher if slower memories are used.

yeungsl, 2017-07-17:
Yes, that is what I assumed. But I did some calculation on memory stall cycles. I found that in one run, the memory stall cycles are higher that what is output by the counter. 

For example, my SST configuration is Ariel with 1 core(2.5 Ghz)
1k L1, 4k L2(accesss latency: 8 cycles), 8K L3(accesss latency: 16 cycles) and 4G Memory(accesss latency: 50ns).

Stat Output:
 a0.cycles.0 : Accumulator : Sum.u64 = 3448896526; SumSQ.u64 = 3448896526; Count.u64 = 3448896526; 
l1cache_0.CacheHits : Accumulator : Sum.u64 = 178401336; SumSQ.u64 = 178401336; Count.u64 = 178401336; 
 l1cache_0.CacheMisses : Accumulator : Sum.u64 = 62004210; SumSQ.u64 = 62004210; Count.u64 = 62004210; 
 l2cache_0.CacheHits : Accumulator : Sum.u64 = 6911541; SumSQ.u64 = 6911541; Count.u64 = 6911541; 
 l2cache_0.CacheMisses : Accumulator : Sum.u64 = 55092669; SumSQ.u64 = 55092669; Count.u64 = 55092669; 
l3cache.CacheHits : Accumulator : Sum.u64 = 1288; SumSQ.u64 = 1288; Count.u64 = 1288; 
 l3cache.CacheMisses : Accumulator : Sum.u64 = 55091381; SumSQ.u64 = 55091381; Count.u64 = 55091381; 

Theoretically:
Memory Stall Cycles = Requests served by L2 * Latency cycles + Requests served by L3 * Latency cycles + Reqeusts served by Memory * Latency cycles

Memory Latency Cycles = access time  / (1/frequency) = 50 ns/ 0.4 ns/cycle = 125 cycles

But if we just calculate Requests served by Memory * Memory Latency cycles = 55091381 * 125 = 6886422625 cycles, which is already larger than what is output by the Ariel cycle counter.

amroawad2, 2017-07-17:
Many of the misses are served in parallel, so the memory stall cycles can overlap across different instructions.

yeungsl, 2017-07-17:
Okay thanks a lot. I was trying to come up with some ways to find the amount of times that this program spends in memory.

amroawad2, 2017-07-17:
Probably best way to do that is to model a hypothetical perfect cache and run the system, then compare it with the realistic memory configuration execution time. Best wishes.

yeungsl, 2017-07-17:
BTW @amroawad2,
One more thing that does not make sense to me is this number:
 a0.no_ops.0 : Accumulator : Sum.u64 = 446587430; SumSQ.u64 = 446587430; Count.u64 = 446587430; 
 a0.instruction_count.0 : Accumulator : Sum.u64 = 240405149; SumSQ.u64 = 240405149; Count.u64 = 240405149;

I thought here no_ops means the number of instructions that do not do any operations. Whereas Instruction_counts means the number of total instructions executed by the cpu? Why instruction_count is smaller than no_ops? Or instructions_counts is not the total number of instructions executed?

Thanks

amroawad2, 2017-07-17:
Yes, it seems like no_ops (uncategorized instructions) are not counted within instruction_counts:

https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/ariel/arielcore.cc#L344

yeungsl, 2017-07-17:
So I am assuming that instructions_counts + no_ops = Total number of instructions executed?

amroawad2, 2017-07-17:
Yes, this should be the case.
FYI, I have just pushed a change request to have the instruction count including no_ops and memory instructions.

yeungsl, 2017-07-17:
Great, is it on the master branches or devel branches?

amroawad2, 2017-07-17:
still to be merged on the devel branch. I think it will take a few hours to do so. Below is the commit: 
https://github.com/sstsimulator/sst-elements/pull/702/files

yeungsl, 2017-07-17:
Perfect, thanks.

nmhamster, 2017-07-25:
Can we close this issue now?

yeungsl, 2017-07-26:
Definitely, Thanks a lot.

closed 2017-07-26 by yeungsl