sstsimulator/sst-elements #710: Motifs run on single node and single rank crash

lzmartinico, 2017-07-21:
1.  __Detailed description of problem or enhancement__
When I try to run any job (even just Init and Fini) using the loadFile or loadCmd scripts on one node with one rank, sst crashes with an _*** Error in sst: double free or corruption (fasttop)_ (when my torus shape is 1) or _***Error in sst: double free or corruption (!prev)_ (when the torus shape is greater than 1). 
The only exception when this doesn't happen is having a torus shape of 1x1; but other shapes defined in this format will cause a crash.
If my job only contains one motif (just Init or just Fini for instance), there is no crash, regardless of torus shape.
 Is the nXn syntax the correct one to use? Is it a problem to use a shape that is larger then the amount of nodes used?
2. __Describe how to reproduce__
Change the NID_LIST in ember/run/configurations/loadfile to 0, and adjust torus shape through ember/run/configurations/testFile.
Change the numNodes and numRanksPerNode fields to 1 in ember/run/configurations/testCmd
3. __What Operating system(s) and versions __
Debian 9
4. __What version of external libraries (Boost, MPI)__
OpenMPI 1.8.8

nmhamster, 2017-07-25:
Hi @lzmartinico - what compiler are you using? We aren't routinely testing on Debian so we will need a little more information. The configuration you have described (Ember with motifs) is run every night in our testing process and is not currently failing so we will need to identify what is going wrong here. Do you get any output from the run at all prior to the double free being reported or is this happening very early in the run? Thanks for reporting the issue.

nmhamster, 2017-07-25:
@lzmartinico - also what version of SST are you running?

lzmartinico, 2017-07-25:
Thank you for the reply, @nmhamster
I am using G++ 6.3.0-18 as backend for mpicxx (OpenMPI)1.8.8, running SST version 7.1.0 (both core and elements).
I turned verbose mode on, and the crash appear to be taking place immediately after the SST core main event loop starts

cindyrella, 2017-07-26:
Hi @nmhamster I also found a similar behavior when I run testFile (and my network is a torus with more than one node) using a loadfile that only contains
 `[JOB_ID] 0`
 `[NID_LIST] 0`
 `[MOTIF] Init`
 `[MOTIF] Fini`
And the same happens if I run any motif in between Init and Fini.

But I do not get any error when I launch 2 jobs, each with only 1 node: i.e.
 `[JOB_ID] 0`
 `[NID_LIST] 0`
 `[MOTIF] Init`
 `[MOTIF] Fini`
 `[JOB_ID] 1`
 `[NID_LIST] 1`
 `[MOTIF] Init`
 `[MOTIF] Fini`

 I am running  in:
 - ubuntu 16.04 LTS  
 - OpenMPI 1.8.8
 - g++ 5.4.0
 - SST-Core version 7.1.0

allevin, 2020-04-03:
No further response from Authors, closing this issue.

closed 2020-04-03 by allevin