sstsimulator/sst-elements #711: Access time for Simple Memory sub system

yeungsl, 2017-07-24:
New Issue for sst-elements
--------------------------

I was running SST with a configuration of 1KB L1, 4KB L2, 64KB L3 and 4 GB memory. I run the exactly same executable on this configuration just with the access_time of memory model changed. But as the access_time gets larger. The change in cycles counts from Ariel CPU is rather random. For example, I changed the access_time from 50ns to 90000ns, but the change in cycle counts from Ariel CPU is very little.  

access_time : 50ns has the output 24647710502 cycles
access_time : 90000ns has the output 24667601684 cycles

I ran multiple benchmarks and they all get fairly high LLC-miss ratio (the above example is about 55%). So I am wondering did anyone else see this kind of problem?

hughes-c, 2017-07-25:
Is this the new cycle count feature that we added last week?

yeungsl, 2017-07-25:
All good right now. I just found out the Parameter's name should be "backend.access_time" rather than "access_time", since I am building on top of the example from Ariel/frontend/simple/examples/stream/runstream.py. Now everything works as expected.

FYI -- in runstream.py the parameter of mem controller is "access_time", which is not adjusting the access latency at all.

Thanks a lot.

hughes-c, 2017-07-25:
It looks like this is indeed a bug in the test scripts. In addition. there's no entry for backend.access_time for MemController in sst-info.

gvoskuilen, 2017-07-25:
The run*.py scripts are older and out of date, we'll go ahead and update and/or remove them. The up-to-date scripts are ariel_ivb.py and ariel_snb.py. In general, to specify parameters for the memory controller's backend subcomponent, prefix the parameter with 'backend.'.

yeungsl, 2017-07-25:
Also @gvoskuilen @hughes-c, I am just wonder why memory controller's network link bandwidth is NULL by default? Can I change it to anything numbers? Moreover, what does "connected to a network" means? How can I adjust the bandwidth from memory to the processors?

gvoskuilen, 2017-07-25:
The memory controller can be directly connected to another component such as a directory or cache (default) or it can sit on a NoC (network connected). To control bandwidth in the network case, you can set 'network_bw'. The format for network_bw is a number with units, such as '20GB/s'. To control bandwidth in the direct connect case, you can limit the number of requests handled each cycle using 'max_requests_per_cycle'. Take a look at the scripts in memHierarchy/tests for examples of different ways to connect memory system components.

yeungsl, 2017-07-25:
Yes regarding to that 'max_requests_per_cycle', should be the number of requests per cycle. The requests size, which is described by 'request_width'. Is 'request_width' in bits or in bytes? Because I think normally it would be in bits.

hughes-c, 2017-07-31:
@yeungsl sst-info reports that the variable expects a size in bytes. You can verify this by checking the MemReq object in membackend/memBackendConvertor.h.

Can I close this issue?

closed 2017-08-02 by hughes-c