sstsimulator/sst-elements #721: MemHierarchy sdl8-1.py Crashes on POWER7 using Threads

nmhamster, 2017-07-27:
Running sdl8-1.py on POWER7 using GCC 4.8.5 with RHEL7 causes a crash when using 2 threads. This runs successfully when using 1 thread.

```/home/sdhammo/sst/20170727-p7/core/bin/sst -n 2 sdl8-1.py```

Output:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x3fffb089f1b0 (LWP 55233)]
0x00000000100d02ac in SST::Simulation::run (this=0x3fffac0008c0) at simulation.cc:530
530	        current_activity->execute();
Missing separate debuginfos, use: debuginfo-install glibc-2.17-78.el7.ppc64 libgcc-4.8.3-9.el7.ppc64 libstdc++-4.8.3-9.el7.ppc64 libtool-ltdl-2.4.2-20.el7.ppc64 python-libs-2.7.5-16.el7.ppc64 zlib-1.2.7-13.el7.ppc64
(gdb) bt
#0  0x00000000100d02ac in SST::Simulation::run (this=0x3fffac0008c0) at simulation.cc:530
#1  0x00000000100638e0 in start_simulation (tid=<optimized out>, info=..., barrier=...) at main.cc:330
#2  0x00000000100640d8 in _M_invoke<0ul, 1ul, 2ul> (this=<optimized out>) at /usr/include/c++/4.8.2/functional:1732
#3  operator() (this=<optimized out>) at /usr/include/c++/4.8.2/functional:1720
#4  std::thread::_Impl<std::_Bind_simple<void (*(unsigned int, std::reference_wrapper<SimThreadInfo_t>, std::reference_wrapper<SST::Core::ThreadSafe::Barrier>))(unsigned int, SimThreadInfo_t&, SST::Core::ThreadSafe::Barrier&)> >::_M_run() (this=<optimized out>) at /usr/include/c++/4.8.2/thread:115
#5  0x00003fffb7b0bed4 in ?? () from /lib64/libstdc++.so.6
#6  0x00003fffb7f0c26c in .start_thread () from /lib64/libpthread.so.0
#7  0x00003fffb77d7440 in .__clone () from /lib64/libc.so.6
```

bjmoor, 2017-07-27:
Another stack of same type of run:

```
cpu: Received Request with command 2 (addr 0x6428c) [Time: 1343] [9 outstanding requests]
cpu: 8084 Issued Read for address 0x3dd8c
FATAL: dirctrl, Error: Received unexpected response from memory - matching request not found. Addr = 0x81540, Cmd = NULLCMD. Time = 227213ns
SST Fatal Backtrace Information:
    0 : /ascldap/users/bjmoor/sst-inst/bin/sst(_ZNK3SST6Output5fatalEjPKcS2_jS2_z+0x218) [0x100c9588]
    1 : /ascldap/users/bjmoor/sst-inst/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy19DirectoryController20handleMemoryResponseEPNS_5EventE+0x6f8) [0x3fffb0ae1f6c]
    2 : /ascldap/users/bjmoor/sst-inst/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST5Event7HandlerINS_12MemHierarchy19DirectoryControllerEvEclEPS0_+0x9c) [0x3fffb0af8e14]
    3 : /ascldap/users/bjmoor/sst-inst/bin/sst(_ZN3SST5Event7executeEv+0x34) [0x100a1af4]
    4 : /ascldap/users/bjmoor/sst-inst/bin/sst(_ZN3SST10Simulation3runEv+0x254) [0x100d18d4]
    5 : /ascldap/users/bjmoor/sst-inst/bin/sst() [0x10064528]
    6 : /ascldap/users/bjmoor/sst-inst/bin/sst(_ZNSt6thread5_ImplISt12_Bind_simpleIFPFvjR15SimThreadInfo_tRN3SST4Core10ThreadSafe7BarrierEEjSt17reference_wrapperIS2_ESB_IS7_EEEE6_M_runEv+0x34) [0x10064e94]
    7 : /lib64/libstdc++.so.6(+0x10bdd4) [0x3fffb7b7bdd4]
    8 : /lib64/libpthread.so.0(+0x8728) [0x3fffb7f48728]
    9 : /lib64/libc.so.6(clone+0x98) [0x3fffb788d210]
[Thread 0x3fffb06ef1b0 (LWP 132572) exited]
[Inferior 1 (process 132564) exited with code 0377]
```

nmhamster, 2017-07-27:
Investigation seems to suggest these are most likely memory pool issues in the core. Closing.

closed 2017-07-27 by nmhamster