sstsimulator/sst-elements #738: Help with simple system using  ring Interconnect

IT302, 2017-08-14:
I am new to sst and struggling  to simulate a simple system  with  a ring interconnect.  Simulation  starts but issues only one memory operation. Could someone please point out what am doing wrong here? A snippet  of the simulation's output is shown below and my configuration is attached.

```
Prospero Component Statistics:
------------------------------------------------------------------------
- Completed at:                          5000000000 ns
- Cycles with ops issued:                1 cycles
- Cycles with no ops issued (LS full):   9999999998 cycles
------------------------------------------------------------------------
- Reads issued:                          0
- Writes issued:                         1
- Split reads issued:                    0
- Split writes issued:                   0
- Bytes read:                            0
- Bytes written:                         8
------------------------------------------------------------------------
- Bandwidth (read):                      0 B/s
- Bandwidth (written):                   1.6 B/s
- Bandwidth (combined):                  1.6 B/s
- Avr. Read request size:                                1.00 bytes
- Avr. Write request size:                               8.00 bytes
- Avr. Request size:                                     8.00 bytes

Prospero Component Statistics:
------------------------------------------------------------------------
- Completed at:                          5000000000 ns
- Cycles with ops issued:                1 cycles
- Cycles with no ops issued (LS full):   9999999998 cycles
------------------------------------------------------------------------
- Reads issued:                          0
- Writes issued:                         1
- Split reads issued:                    0
- Split writes issued:                   0
- Bytes read:                            0
- Bytes written:                         8
------------------------------------------------------------------------
- Bandwidth (read):                      0 B/s
- Bandwidth (written):                   1.6 B/s
- Bandwidth (combined):                  1.6 B/s
- Avr. Read request size:                                1.00 bytes
- Avr. Write request size:                               8.00 bytes
- Avr. Request size:                                     8.00 bytes
```

[sst-config.txt](https://github.com/sstsimulator/sst-elements/files/1222142/sst-config.txt)

nmhamster, 2017-08-14:
Hi @IT302 - can you check the size of your trace files? How did you obtain the traces?

IT302, 2017-08-14:
@nmhamster
I don't think  the trace files  are problem  they are 56 MiB.  Simulation works fine with the same trace files when using a single router to connect the components. 

I have changed the configuration (attached) to use "memHierarchy.trivialCPU" and it  still doesn't work. The output from the simulator is also attached. 

Many thanks for your help.

[sst-ouput.txt](https://github.com/sstsimulator/sst-elements/files/1222792/sst-ouput.txt)
[sst-config-trivial-cpu.txt](https://github.com/sstsimulator/sst-elements/files/1222759/sst-config-trivial-cpu.txt)

gvoskuilen, 2017-08-14:
@IT302 The merlin torus topology needs to be connected in both positive and negative directions. If you only want it connected in one, you can use a mesh topology instead.

IT302, 2017-08-15:
Many thanks @gvoskuilen 

My connections looks like this: 

> link_torus_0_1 = sst.Link("link_torus_0_1")
link_torus_1_2 = sst.Link("link_torus_1_2")
link_torus_2_0 = sst.Link("link_torus_2_0")
link_torus_0_1.connect((comp_torus0,"port0","2000ps"), (comp_torus1,"port1","2000ps"))
link_torus_1_2.connect((comp_torus1,"port0","2000ps"), (comp_torus2,"port1","2000ps"))
link_torus_2_0.connect((comp_torus2,"port0","2000ps"), (comp_torus0,"port1","2000ps"))

What am I missing here?

gvoskuilen, 2017-08-15:
@IT302 Each router needs both ports 0 & 1 connected, one in each direction:
```
link_torus_0_1 = sst.Link("link_torus_0_1")
link_torus_0_2 = sst.Link("link_torus_0_2")
link_torus_1_2 = sst.Link("link_torus_1_2")
link_torus_1_0 = sst.Link("link_torus_1_0")
link_torus_2_0 = sst.Link("link_torus_2_0")
link_torus_2_1 = sst.Link("link_torus_2_1")
link_torus_0_1.connect( (comp_torus0, "port0", "2000ps"), (comp_torus1, "port1", "2000ps") )
link_torus_0_2.connect( (comp_torus0, "port1", "2000ps"), (comp_torus2, "port0", "2000ps") )
link_torus_1_2.connect( (comp_torus1, "port0", "2000ps"), (comp_torus2, "port1", "2000ps") )
link_torus_1_0.connect( (comp_torus1, "port1", "2000ps"), (comp_torus0, "port0", "2000ps") )
link_torus_2_0.connect( (comp_torus2, "port0", "2000ps"), (comp_torus0, "port1", "2000ps") )
link_torus_2_1.connect( (comp_torus2, "port1", "2000ps"), (comp_torus1, "port0", "2000ps") )
```

feldergast, 2017-08-15:
I took a look at the configuration.  Your torus links are actually connected correctly.  However, it looks like a couple of parameters are set up wrong for how the network is wired together.  You have the L2cache wired to router 0, and a directory controller connected to routers 1 and 2.  However, the "network_address" parameter for your L2 cache is set to 1 (which means it expects to be wired to router 1) and the directory controller connected to router 1 is give a "network_address" of 0 (which means it expects to be connected to router 0).  If you switch these two parameters, then things should work as expected.

So, set "network_address" to 0 for the component named "l2cache" and set "network_address" to 1 for the component named "dirctrl0".

IT302, 2017-08-16:
Switching  the parameters  fixed it.  

Thanks to everyone for your help.

closed 2017-08-16 by IT302