sstsimulator/sst-elements #759: Ariel gives "Error: Ariel did not understand command" while executing a bash script

shavvn, 2017-09-05:
Hi there,

I was trying to use ariel to execute a bash script when it failed and gives an error of "Error: Ariel did not understand command".

**Steps to reproduce:**

1. create a simple bash script with 3 lines as *test.sh*

```
echo "this is ok"
ls
echo "this is not"
```

2. change the sst-elements/src/sst/elements/ariel/frontend/simple/examples/stream/ariel_ivb.py to execute the script with the following lines:

```
"executable"          : "bash",
"apparg0"             : "test.sh",
"appargcount"         : "1",
```

3. Run `sst ariel_ivb.py`, get the output:

```
this is ok
FATAL: ArielComponent[arielcore.cc:329:refillQueue] Error: Ariel did not understand command (32) provided during instruction queue refill.
FATAL: ArielComponent[arielcore.cc:329:refillQueue] Error: Ariel did not understand command (32) provided during instruction queue refill.
```

I can successfully run those commands individually or run multiple *echo* commands in a script but as long as I put *ls* in the script it will break, any idea how to fix this? 

**system info**:
sst version 7.1.0
pin version 2.14
system Ubuntu 14

Thanks

nmhamster, 2017-09-05:
@shavvn - thanks for reporting the issue. Command 32 represents the start of an instruction being provided to Ariel so this should be OK. Judging by the location in the core (from your output) it looks like Ariel is attempting to issue a second instruction while the first one has not closed yet. We will need to look into that but it sounds like `bash` might be doing something bad here. If we provide a fix in the `devel` version of SST will this still be useful for you?

shavvn, 2017-09-05:
@nmhamster yeah definitely, I'm okay using a devel version. 

Thank you for your help and prompt response.

allevin, 2020-04-03:
No further response from author, closing this issue.

closed 2020-04-03 by allevin