sstsimulator/sst-elements #782: Is it possible to use firefly and merlin only?

fisher5090, 2017-09-25:
I dont want to use MPI-like programming to drive the simulator ( I focus on the network), Is there any way for me to inject packets to firefly directly? Ember is too complicated to relate MPI primitive to packet in firefly::nic.

It is not an issue, but really a question I want to know. I have spent 3 weeks on the source code of merlin, firefly and ember, a nightmare without enough documents, especially ember.

nmhamster, 2017-11-21:
@fisher5090 - thanks for the feedback. You can generate packets directly into Merlin to simulate network behavior but this will not have the MPI semantics you might want. It sounds like you want to be able to map MPI operations directly to packets which really requires some fairly complex handling in order to be accurate (hence the complexity). Let us know if a deep dive on the internals is useful and we can see what we can do for you.

allevin, 2020-04-03:
No further response from author, closing this issue.

closed 2020-04-03 by allevin