sstsimulator/sst-elements #789: Ariel will not compile on High Sierra

jpvandy, 2017-10-10:
This may be moot, since PIN won't work with High Sierra.

The file that won't compile is fesimple.cc
Here is the compile attempt:
/usr/bin/clang++ -O3 -shared \
	-g -O2  -std=c++1y -D__STDC_FORMAT_MACROS  \
	-I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7  -I/Users/sstbuild/jenkins/workspace/SST__Nightly_MacOSX_10.13_OMPI_1.8_Boost_1.56_clang_900.0.37_mainline/1_2017-10-09_11-36-58/local/sst-core/include/sst/core -I/Users/sstbuild/jenkins/workspace/SST__Nightly_MacOSX_10.13_OMPI_1.8_Boost_1.56_clang_900.0.37_mainline/1_2017-10-09_11-36-58/local/sst-core/include \
	 \
	-DBIGARRAY_MULTIPLIER=1 \
	-DTARGET_IA32E -DHOST_IA32E -fPIC -DTARGET_MAC \
	-I./ \
	-I/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/source/include/pin \
	-I/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/ \
	-I/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/extras/components/include \
	-I/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/source/include/pin/gen/ \
	-I/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/extras/xed-intel64/include \
	-I../../../../src/sst \
	-I../../../../src/sst/elements/ariel \
	-fomit-frame-pointer -fno-stack-protector \
	-Wl,-exported_symbols_list \
	-Wl,/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/source/include/pin/pintool.exp \
	-L/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/intel64/lib \
	-L/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/intel64/lib-ext \
	-L/usr/local/module-pkgs/pin/pin-2.14-71313-clang.5.1-mac/extras/xed-intel64/lib \
	-o fesimple.dylib ../../../../src/sst/elements/ariel/frontend/simple/fesimple.cc \
	-stdlib=libstdc++ \
	-lpin -lxed -lpindwarf -lpthread  -lz
clang: warning: libstdc++ is deprecated; move to libc++ [-Wdeprecated]
1 warning generated.
mv -f .deps/libariel_la-arielcpu.Tpo .deps/libariel_la-arielcpu.Plo
1 warning generated.
mv -f .deps/libariel_la-arielcore.Tpo .deps/libariel_la-arielcore.Plo
../../../../src/sst/elements/ariel/frontend/simple/fesimple.cc:727:23: error: ordered comparison between pointer and zero ('UINT64 *' (aka 'unsigned long long *') and 'int')
    if(lastMallocSize >= 0) {
       ~~~~~~~~~~~~~~ ^  ~
1 error generated.

jleidel, 2018-01-12:
@jpvandy what version of the OSX tool chain are you running?

hughes-c, 2020-01-26:
@jpvandy Is this still a problem?

allevin, 2020-04-03:
@jpvandy is this still an issue?

jpvandy, 2020-04-03:
being fixed on Mojave for Rel 10

closed 2020-04-03 by jpvandy