sstsimulator/sst-elements #800: SST Core is not finding a symbol

amroawad2, 2017-11-04:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

Running SST even with simpleElementExample is failing with this message:
	/home/RESEARCH/SST/BUILD_ELEMENTS/lib/sst-elements-library/libsimpleElementExample.so: undefined symbol: _ZNK3SST9Component31doesComponentInfoStatisticExistERKSs
Could not find ELI block simpleElementExample_eli in simpleElementExample
FATAL:  SST Core: can't find requested component simpleElementExample.simpleLookupTableComponent.
 FATAL:  SST Core: can't find requested component simpleElementExample.simpleLookupTableComponent.
 SST Fatal Backtrace Information:
    0 : ../BUILD_CORE/bin/sst(_ZNK3SST6Output5fatalEjPKcS2_jS2_z+0x40a) [0x4bb3ba]
    1 : ../BUILD_CORE/bin/sst(_ZN3SST7Factory15CreateComponentEmRNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_6ParamsE+0x5fd) [0x4a565d]
    2 : ../BUILD_CORE/bin/sst(_ZN3SST10Simulation13performWireUpERNS_11ConfigGraphERKNS_8RankInfoEm+0x3be) [0x4c355e]
    3 : ../BUILD_CORE/bin/sst() [0x465f94]
    4 : ../BUILD_CORE/bin/sst(main+0x1865) [0x45a115]
    5 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f8b0dee4830]
    6 : ../BUILD_CORE/bin/sst(_start+0x29) [0x465999]

2 - Describe how to reproduce

I am using Ubuntu 16.04, gcc-4.8, and the devel version of SST at the time of filing this issue. I use pin-2.14, even though not used in this test. In this specific test, which fails, I use:

../BUILD_CORE/bin/sst ../sst-elements/src/sst/elements/simpleElementExample/tests/test_simpleLookupTable.py 

3 - What Operating system(s) and versions 

Ubuntu 16.04

4 - What version of external libraries (Boost, MPI)

No boost. OpenMPI 1.10

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

amroawad2, 2017-11-07:
@allevin Hi Arun, does the issue look to be familiar?

allevin, 2017-11-07:
Hi Amro,  I spoke to Si about this.  Not sure why you would be having this issue.  The only thing I know is that PIN does not work on the newer platforms (Ubuntu 16).  You might try building without PIN and see if that improves.  The other possibility is that your elements are not being properly registered.  I would recommend that you build core and elements in a new separate clean directory, and make sure that you don't have any paths to older versions.  When the core is launched, it looks at the sstsimulator.conf file in `<sst-core>/etc/sst` to identify where the files are located.  If you just build sst-core by itself, you should be able to run sst-info and get a almost empty data set.  If you see that are not yet built, then something is wrong in the .conf file.

amroawad2, 2017-11-07:
Thanks alot @allevin , I will check that and see how it goes.

nmhamster, 2017-11-20:
@amroawad2 - any news on whether building without PIN makes this issue go away?

amroawad2, 2017-11-21:
Yes this solved with removing and reinstalling again.

closed 2017-11-21 by amroawad2