sstsimulator/sst-elements #807: Allreduce does not work with a subcommunicator

chabbimilind, 2017-11-19:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

enQ_allreduce() on a subcommunicator crashes.

2 - Describe how to reproduce

Use this code in a motif:

// assumes m_layerComm is a Communicator class member
bool EmberBugGenerator::generate( std::queue<EmberEvent*>& evQ )
{
    if( 0 == m_loopIndex) {
        verbose(CALL_INFO, 1, 0, "rank=%d size=%d\n", rank(),size());
        }
    if (  m_loopIndex ==  0 ) {
                enQ_commSplit( evQ, GroupWorld,  rank()%2,  rank()/2, &m_layerComm );
    }
    // The below line causes a crash. 
    enQ_allreduce(evQ, NULL, NULL, 1, FLOAT, SUM, m_layerComm);

    if ( ++m_loopIndex == iterations ) {
        return true;
    } else {
        return false;
    }
}

3 - What Operating system(s) and versions 
DISTRIB_DESCRIPTION="Ubuntu 14.04.2 LTS"
NAME="Ubuntu"
VERSION="14.04.2 LTS, Trusty Tahr"
4 - What version of external libraries (Boost, MPI)
OpenMPI

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
7.1.0 public release

6 - Fill out Labels, Milestones, and Assignee fields as best possible

nmhamster, 2017-11-20:
@chabbimilind - we are looking into this with the development version (@mjleven).

mjleven, 2017-11-22:
Pull request 817 fixed this

closed 2017-11-22 by mjleven