sstsimulator/sst-elements #836: sst_config.h not found while building

deanchester, 2017-11-29:
I am building SST on macOS 10.13.1 (17B48) with Xcode, and I am running in to an issue building SST Elements. 

I am following the detailed instructions on the SST website. 

[Attached](https://github.com/sstsimulator/sst-elements/files/1513617/SSTCoreConfig.txt) is the output from the configure command for the SST-core. 

[Attached](https://github.com/sstsimulator/sst-elements/files/1513640/SSTElementsConfig.txt) is the output from the configure command for SST elements. 

Here is some additional output:

    deangchester@Deans-MBP:SST/sst-elements-library-7.1.0 $ ./configure -- prefix=$SST_ELEMENTS_HOME --with-sst-core=$SST_CORE_HOME >> SSTElementsConfig.txt
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'
    SST: Unable to open configuration file 'NONE/etc/sst/sstsimulator.conf'

Here is the output from `make all`

     deangchester@Deans-MBP:SST/sst-elements-library-7.1.0 $ make all
     Making all in src
    /Library/Developer/CommandLineTools/usr/bin/make  all-recursive
     Making all in sst
     Making all in elements
     Making all in cacheTracer
      CXX      cacheTracer.lo
     cacheTracer.cc:16:10: fatal error: 'sst_config.h' file not found
    #include "sst_config.h"
             ^~~~~~~~~~~~~~
     1 error generated.
     make[5]: *** [cacheTracer.lo] Error 1
     make[4]: *** [all-recursive] Error 1
     make[3]: *** [all-recursive] Error 1
     make[2]: *** [all-recursive] Error 1
     make[1]: *** [all] Error 2
     make: *** [all-recursive] Error 1

allevin, 2017-11-29:
@deanchester The elements package requires sst-core to be compiled, installed and functional before building sst-elements.

Here are a few questions/suggestions:

1. Did you run make install on the sst-core?  
1. Can you run `sst --version` before configuring/compiling sst-elements.  
1. Can you run `sst-info` before configuring/compiling sst-elements.  
1. Can you verify that the sst applications you are running are the ones that you compiled (`which sst`)?
1 Can you try configuring and building sst-core without Boost?
1. Can you provide us with your command lines used when configuring sst-core along with the paths to $SST_ELEMENTS_HOME and $SST_CORE_HOME?
1. Your config command line for sst-elements seems to have an error for prefix `-- prefix` should be `--prefix`.

The summary at the end of configuration for sst-elements indicates that the `--with-sst-core` is undefined.  The config file seems to be finding sst-core applications in `/usr/local/bin`.  Are you sure that your environment variable `$SST_CORE_HOME` is populated with the correct path?

```
checking for strlcpy... yes
checking whether time.h and sys/time.h may both be included... (cached) yes
checking for sst-config... /usr/local/bin/sst-config
checking for sst-config tool... found /usr/local/bin/sst-config
checking for sst-register... /usr/local/bin/sst-register
checking for sst-register tool... found /usr/local/bin/sst-register
checking for Mac OSX... yes
checking for pin.sh... no
```

deanchester, 2017-11-29:
Thanks for the help. `which sst` was pointing to some random system location removing that has fixed the issue. It was from when I didn't previously set a local path for installing sst.

closed 2017-11-29 by deanchester