sstsimulator/sst-elements #840: Simulator component with trace file

liuyidinguh, 2017-12-05:
New Issue for sst-elements
--------------------------
I have some problems about trace file (related to jobs instead of memory trace) generation in SST for a specific applications with using default scheduler to create a SDL file. I use Ariel-based core component, and I wonder for different configurations with a specific application,  the trace file(related to assign jobs ) will be same and how I generate the trace file needed. I noticed the trace file can be generated from Parallel Workloads format, but I have no idea how to relate an application with this format in different configuration in SST. I'll appreciate it if you could help me to solve that. Thanks!

nmhamster, 2018-02-19:
@liuyidinguh - sorry for the long wait for us to reply. Would you rather generate trace files outside of Ariel? We have independent PIN-based tools which are much easier to run. Rather than capture these using Ariel, you might want to use Prospero (which is designed for trace reply). There is a tracing tool based on PIN in src/sst/elements/prospero/tracetool. This can capture trace files that you can then load for reply into Prospero.

hughes-c, 2018-09-20:
@liuyidinguh Did this resolve your problem?

liuyidinguh, 2018-09-20:
Thanks for replying. I have solved it by using an external component MacSim
with Pintool.

Best,

Yiding

closed 2018-09-20 by hughes-c