sstsimulator/sst-elements #857: The error happens when I use Ariel and Pintool

liuyidinguh, 2017-12-18:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
When I want to use Ariel and pintool, the error shows:
FATAL: ArielComponent[arielcpu.cc:461:forkPINChild] Launching trace child failed!  Child Exited with status 255
FATAL: ArielComponent[arielcpu.cc:461:forkPINChild] Launching trace child failed!  Child Exited with status 255
 I have tried to use both the example of Pintool and Pintool I wrote, which have been successfully compiled with the GCC 4.9.2, Pin 2.14 and SST/6.1.0.
I think  it may be resulted from the injection failure between Pintool parent process and child application process.
2 - What Operating system(s) and versions 
GNU/Linux bash-4.1

Thanks for helping me!

nmhamster, 2017-12-18:
Hi thanks for reporting this issue. Few questions:

Does your application run outside of Ariel successfully? 

Can you "ldd" on your binary and reply with the output to this issue? This helps us identify possible library conflicts

Also, can we get the Ariel section of your Python input to SST?

Thanks! We won't be able to look at this until a little later this week.

liuyidinguh, 2017-12-18:
Yes, the application runs successfully outside, and when I use the example of application in SST(stream), the error is same.

In the second problem, what is " the "ldd" on my binary"? Sorry that I'm confused about that.

Here is the Ariel section configuration:
import sst
import sys
import ConfigParser, argparse
from utils_ariel import *
import os

sst_root = os.getenv( "HOME" )
corecount = 2
# Parse commandline arguments
parser = argparse.ArgumentParser()
parser.add_argument("-c", "--config", help="specify configuration file", required=True)
parser.add_argument("-v", "--verbose", help="increase verbosity of output", action="store_true")
parser.add_argument("-s", "--statfile", help="statistics file", default="./stats.csv")
parser.add_argument("-l", "--statlevel", help="statistics level", type=int, default=16)

args = parser.parse_args()

verbose = args.verbose
cfgFile = args.config
statFile = args.statfile
statLevel = args.statlevel

# Build Configuration Information
config = Config(cfgFile, verbose=verbose)

print "Configuring Network-on-Chip..."

router = sst.Component("router", "merlin.hr_router")
router.addParams(config.getRouterParams())
router.addParam('id', 0)

# Connect Cores & caches
cpu = sst.Component("ariel8", "ariel.ariel")
cpu.addParams( {
                "verbose" : "0",
                "maxcorequeue" : "256",
                "maxissuepercycle" : "2",
	              "pipetimeout" : "0",
	              "corecount" : str(corecount),
               # "profilefunctions": "1",
 	              "executable" :  sst_root + "/sst/Experiment/ariel/bfs",
	              "appargcount" : "2",
                "apparg0" : "--dataset",
                "apparg1" :  sst_root + "/sst/Experiment/ariel/dataset/small",
 	              "arielmode" : "1",
                #"arieltool": sst_root + "/sst/sst-tools-devel/tools/ariel/fesimple/fesimple.so" ,
                "arieltool" : sst_root + "/sst/Experiment/Pintool/task.so",
                "launchparamcount":  "0" ,
                #"launchparam0": "-injection",
                #"launchparam1": "self" ,
                #"memorylevels" : "3",
	       #"arielinterceptcalls" : "0"
                # "defaultlevel" : "0")
                               	} )
for next_core_id in range(0,corecount):
    print "Configuring core %d..."%next_core_id

    l1 = sst.Component("l1cache_%d"%(next_core_id), "memHierarchy.Cache")
    l1.addParams(config.getL1Params())

    l2 = sst.Component("l2cache_%d"%(next_core_id), "memHierarchy.Cache")
    l2.addParams(config.getL2Params())
    
    
    l3 = sst.Component("l3cache_%d"%(next_core_id), "memHierarchy.Cache")
    l3.addParams(config.getL3Params())
    l3.addParam( "network_address", next_core_id )
    l3.addParam( "slice_id", next_core_id )
    
    connect("cpu_cache_link_%d"%next_core_id,
            cpu, "cache_link_%d"%next_core_id,
            l1, "high_network_0",
            config.ring_latency).setNoCut()

    connect("l2cache_%d_link"%next_core_id,
            l1, "low_network_0",
            l2, "high_network_0",
            config.ring_latency).setNoCut()
    
    connect("l3cache_%d_link"%next_core_id,
            l2, "low_network_0",
            l3, "high_network_0",
            config.ring_latency).setNoCut()
  
    connect("l3_ring_link_%d"%next_core_id,
            l3, "directory",
            router, "port%d"%next_core_id,
            config.ring_latency)

# Connect Memory and Memory Controller to the ring
mem = sst.Component("memory", "memHierarchy.MemController")
mem.addParams(config.getMemParams())

dc = sst.Component("dc", "memHierarchy.DirectoryController")
dc.addParams(config.getDCParams(0))
dc.addParam("network_address", str(corecount))

connect("mem_link_0",
        mem, "direct_link",
        dc, "memory",
        config.ring_latency)

connect("dc_link_0",
        dc, "network",
        router, "port%d"%corecount,
        config.ring_latency)

# ===============================================================================

# Enable SST Statistics Outputs for this simulation
sst.setStatisticLoadLevel(statLevel)
sst.enableAllStatisticsForAllComponents({"type":"sst.AccumulatorStatistic"})

sst.setStatisticOutput("sst.statOutputCSV")
sst.setStatisticOutputOptions( {
    "filepath"  : statFile,
    "separator" : ", "
    } )

print "Completed configuring the multicore model"

hughes-c, 2020-01-26:
@liuyidinguh Are you still having a problem with this?

allevin, 2020-04-03:
No further response from author, closing this issue.

closed 2020-04-03 by allevin

juddyoung, 2020-05-18:
@hughes-c 
hello Hughes, I also met 'FATAL: ArielComponent[arielcpu.cc:512:forkPINChild] Launching trace child failed!  Child Exited with status 255', the following is my configuration info.

ariel_params={
        "verbose": "0",
        "maxcorequeue": "256",
        "maxissuepercycle": "2",
        "pipetimeout": "10",
        "corecount": groups * cores_per_group,
        #canneal

"arieltool" : "../pin-2.14-71313-gcc.4.4.7-linux/source/tools/ManualExamples/obj-intel64/pinatrace.so",
"launcher" : "../pin-2.14-71313-gcc.4.4.7-linux/pin",
"instrument_instructions" : "1",
        "executable" : "../parsec-3.0/pkgs/kernels/canneal/src/canneal",
"launchparamcount" : 0,
#"launchparam0" : "-ifeellucky",
        "appargcount" : 5,
        "apparg0" : "1",#NTHREADS NSWAPS TEMP NETLIST
        "apparg1" : "100",
        "apparg2" : "5",
        "apparg3" : "../parsec-3.0/pkgs/kernels/canneal/inputs/10.nets",
        "apparg4" : "32",
#canneal

        "arielmode" : "1",#"Tool interception mode, set to 1 to trace entire program (default), set to 0 to delay tracing until ariel_enable() call., set to 2 to attempt auto-detect", "2"
"max_insts" : 500000000,
"clock":str(clock),#add by judd
        "memmgr.memorylevels" : "2",
        "arielinterceptcalls" : "0",
        "memmgr.defaultlevel" : "0",
"checkaddresses":1#add by judd
}
ariel = sst.Component("a0", "ariel.ariel")
ariel.addParams(ariel_params)