sstsimulator/sst-elements #896: Help to Run a motif

RaviTejaJp, 2018-01-31:
Can someone please explain how to run a motif?

os : ubuntu 16.04
version : sst 7.X
Firstly i am new to using SST i downloaded sstcore, elements 
now i am looking at givien motifs and py files.
i want a help to run motif file so plz provide it.

if you have a handbook on using sst (a little more detailed on how to create a new motif and run it in newly simulated hardware plz provide it.) 
Thank you.
Even a little help is more appreciated.
i am struck here since a week because of not having any proper documentation in running motif.

nmhamster, 2018-02-19:
@RaviTejaJp - what motif do you want to run? Are you wanting to run a network simulation or something with memory subsystem analysis? Thanks!

RaviTejaJp, 2018-02-20:
Actually i am learning SST and trying to run any kind of motif and
understand its functionality and modify motif and write new ones etc.
So if you can give me some source about motifs which talk about them little
bit deeply it would be very much helpful.

if you don't have any documentation about it then plz explain how to run a
motif of samba for instance.

Thank you.

allevin, 2020-04-03:
No further response from author, closing this issue.

closed 2020-04-03 by allevin