sstsimulator/sst-elements #926: Ariel element crashes for sst7.2/sst-master

adityamdeshpande, 2018-02-27:
When using Ariel component, I get following error. 
FATAL: ArielComponent[arielcpu.cc:475:forkPINChild] Launching trace child failed!  Child Terminated With Signal 11; Core Dump File Created = 128

I thought the problem might be with sst7.2, tried using sst6.1, getting same problem. It only occurs when using Ariel. Seems something related to forking process for the application benchmark in Ariel. I tried running sst6.1 with a previous trace using prospero, seems to be working.

I am using RHEL7.4, Kernel 3.10.0-693.17.1.el7.x86_64
GCC: 4.8.5 
Pin: pin-2.14-71313-gcc.4.4.7-linux

A similar error was reported with on sst simulator github page, albeit no solution suggest
https://github.com/sstsimulator/sst-elements/issues/857

[ariel_error.log](https://github.com/sstsimulator/sst-elements/files/1763551/ariel_error.log)

**I also checked the /var/log from the machine, it reported following problem with pinbin.
type=ANOM_ABEND msg=audit(1519692280.177:456): auid=6813 uid=0 gid=0 ses=22 pid=20382 comm="pinbin" reason="memory violation" sig=6**

Any way to solve the problem? I am planning to downgrade the kernel and test again. Will report the results for that soon.

Here's transcript of the log from sst simulation.
A: Source/pin/injector_nonmac/auxvector.cpp: CopyAux: 291: unexpected AUX VEC type 26

################################################################################
## STACK TRACE
################################################################################
addr2line -C -f -e "/nfs/div1/adeshpan/packages/pin-2.14-71313-gcc.4.4.7-linux/intel64/bin/pinbin" 0x560acd6227f9 0x560acd6235ce 0x560acd6238a0 0x560acd845b3c 0x560acd84b2ff 0x560acd84ce2d 0x560acd83cfe1 0x560acd841b2d 0x560acd83222e 0x560acd83c230 0x7f1b3c9e9c05 0x560acd61717a
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
??
??:0
Detach Service Count: 1
Pin 2.14
Copyright (c) 2003-2015, Intel Corporation. All rights reserved.
@CHARM-VERSION: $Rev: 71293 $
@CHARM-BUILDER: BUILDER
@CHARM-COMPILER: gcc 4.4.7
@CHARM-TARGET: ia32e
@CHARM-CFLAGS:  __OPTIMIZE__=1  __NO_INLINE__=__NO_INLINE__
C: Injector exited with signal 6
E: Wait for injector failed: No child processes
FATAL: ArielComponent[arielcpu.cc:475:forkPINChild] Launching trace child failed!  Child Terminated With Signal 11; Core Dump File Created = 128
FATAL: ArielComponent[arielcpu.cc:475:forkPINChild] Launching trace child failed!  Child Terminated With Signal 11; Core Dump File Created = 128
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD 
with errorcode 1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_jS2_z+0x247) [0x4b4987]
    1 : /nfs/div1/adeshpan/packages/sst7.2/sst-elements7.2/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU12forkPINChildEPKcPPcRSt3mapISsSsSt4lessISsESaISt4pairIKSsSsEEE+0x46e) [0x7feb5128044e]
    2 : /nfs/div1/adeshpan/packages/sst7.2/sst-elements7.2/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU4initEj+0x9b) [0x7feb512806ab]
    3 : sst(_ZN3SST10Simulation10initializeEv+0x64) [0x4ba614]
    4 : sst() [0x463b02]
    5 : sst(main+0xef3) [0x4571f3]
    6 : /lib64/libc.so.6(__libc_start_main+0xf5) [0x7feb6190ec05]
    7 : sst() [0x46355e]

nmhamster, 2018-03-11:
@adityamdeshpande - this error indicates that the child process may have died with a memory error. Does the code run successfully outside of Ariel if you run it multiple times?

adityamdeshpande, 2018-03-12:
yes it works outside of ariel. I think the problem is with pin and kernel. For some reason with this kernel version, pin doesn't work.

nmhamster, 2018-05-25:
@adityamdeshpande - have you tried this with any recent updates? Ariel has had a few bug fixes added now.

adityamdeshpande, 2018-05-29:
@gvoskuilen @nmhamster : I had tried adding -ifeellucky to arielcpu.cc, didn't work. BTW this problem is only with RHEL7/ CentOS7 with kernel above 3.0. This is working with CentOS6.9 which uses Kernel 2.62. In order to keep simulator working, I downgraded my machine to CentOS6.9 to generate traces using Ariel. Not the most elegant one, but works. 
I will check the recent updates to see if they help.

hughes-c, 2018-05-29:
@adityamdeshpande What happened when you added the flag?

adityamdeshpande, 2018-05-29:
@hughes-c: flag was in wrong place. Now added in correct location.  Works now. I will try recent updates soon.

derrickgreenspan, 2018-05-29:
@adityamdeshpande  - where in the code did you place the flag? I am getting this exact error - and when trying to run with -ifeellucky added in pin.sh, the program hangs forever on our system which updated to CentOS7.4.

adityamdeshpande, 2018-05-30:
In arielcpu.cc file, I did two changes.
// Added +1 for additional launch parameter
const uint32_t pin_arg_count = 25 + launch_param_count +1;  

After line: execute_args[arg++] = const_cast<char*>("-follow_execv");
Add:  execute_args[arg++] = const_cast<char*>("-ifeellucky");

derrickgreenspan, 2018-05-30:
@adityamdeshpande and @hughes-c : thank you.

If you are using the latest version of SST from the master branch, with my changes which stall the CPU core during ariel_fence() instructions, the program hangs and acts very weirdly with -ifeellucky. I know @VamseeReddyK can reproduce this, but can anyone else? 

I may have to simply set up a VM with CentOS 6.9 and do my research that way, unfortunately.

adityamdeshpande, 2018-05-30:
@derrickgreenspan: I am using sst-7.2 release instead of using master branch.

derrickgreenspan, 2018-05-30:
@adityamdeshpande - okay, if you use master branch, it will likely give the same issue. I'm not sure what has changed between release and master beyond my changes to ariel (ariel_fence() and ariel_flush() support).

I'm not really sure what to do now... I wish Intel PIN 2.14 still worked properly...

hughes-c, 2018-05-30:
@derrickgreenspan We need more information (and maybe separate issue ticket) to help you. I am unable to reproduce what you describe using the configurations that you sent me. Please provide your environment and, if you think it's stalled, a stack trace of where it's occurring.

derrickgreenspan, 2018-05-30:
@hughes-c  :

With CentOS 7.4, kernel build 3.10.0-693.el7.x86_64, intel pin, without the -ifeellucky flag, crashes with the following:

FATAL: ArielComponent[arielcpu.cc:517:forkPINChild] Launching trace child failed!  Child Terminated With Signal 11; Core Dump File Created = 128
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_jS2_z+0x2aa) [0x4b80da]
    1 : /apps/sst/sst-7.2.0-openmpi-1.8.3-gcc-4.9.2-master/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU12forkPINChildEPKcPPcRSt3mapISsSsSt4lessISsESaISt4pairIKSsSsEEE+0x435) [0x2b9680bc53d5]
    2 : /apps/sst/sst-7.2.0-openmpi-1.8.3-gcc-4.9.2-master/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU4initEj+0xc6) [0x2b9680bc5676]
    3 : sst(_ZN3SST10Simulation10initializeEv+0x77) [0x4bdcf7]
    4 : sst() [0x46472a]
    5 : sst(main+0xdb9) [0x457439]
    6 : /lib64/libc.so.6(__libc_start_main+0xf5) [0x2b9671cd2c05]
    7 : sst() [0x464060]

The stack trace is found [here](https://github.com/sstsimulator/sst-elements/files/2055720/stacktrace_no_lucky.txt)

When the -ifeellucky flag is passed, there is no stacktrace produced, rather, from @VamseeReddyK 's observations: "It is hanging on maximum pending transaction. The requests are not getting served fast and due to this Ariel core reaches maximum of the pending transactions and does not serve any other new requests". The output of one such run is attached [here](https://github.com/sstsimulator/sst-elements/files/2055727/output_ifeellucky.txt).

hughes-c, 2018-05-31:
@derrickgreenspan @VamseeReddyK I still can't reproduce the stall. I have tried on:
3.10.0-693, gcc 4.8.5, Python 2.7.14  
3.10.0-229, gcc 4.9.3, Python 2.7.13  
3.10.0-514, gcc 4.9.3/4.8.5, Python 2.7.5

derrickgreenspan, 2018-05-31:
@hughes-c  - this is with the Master branch, correct?
We are using gcc 4.9.2, maybe 4.8.5 with that kernel will work? I will continue to investigate.

derrickgreenspan, 2018-06-01:
@hughes-c 

I went ahead and debugged this some more:
If I just call ariel_flush(), it works fine when the -ifeellucky flag is passed. If, however, I call ariel_fence(), it hangs on SSTARIEL: Execution completed, shutting down.

hughes-c, 2018-06-02:
Closing this since it's getting off-topic.

closed 2018-06-02 by hughes-c

juddyoung, 2020-05-17:
> @adityamdeshpande - this error indicates that the child process may have died with a memory error. Does the code run successfully outside of Ariel if you run it multiple times?

hello sir, I also met the error when I launch the PIN in ariel, following is my code, can you tell me anything wrong?
ariel_params={
        "verbose": "0",
        "maxcorequeue": "256",
        "maxissuepercycle": "2",
        "pipetimeout": "10",
        "corecount": groups * cores_per_group,

	#"arieltool" : "/home/judd-young/scratch/src/pin-2.14-71313-gcc.4.4.7-linux/source/tools/ManualExamples/obj-intel64/pinatrace.so",
	#"launcher" : "/home/judd-young/scratch/src/pin-2.14-71313-gcc.4.4.7-linux/pin",
	"arieltool" : "/home/judd-young/PIN-3.13/pin-3.13-98189-g60a6ef199-gcc-linux/source/tools/ManualExamples/obj-intel64/pinatrace.so",
	"launcher" : "/home/judd-young/PIN-3.13/pin-3.13-98189-g60a6ef199-gcc-linux/pin",
        "executable" : "/home/judd-young/Desktop/parsec-3.0/pkgs/kernels/canneal/src/canneal",
	"instrument_instructions" : "1",
	"launchparamcount" : 1,
	"launchparam0" : "-ifeellucky",
        "appargcount" : 5,
        "apparg0" : "1",#NTHREADS NSWAPS TEMP NETLIST
        "apparg1" : "100",
        "apparg2" : "5",
        "apparg3" : "/home/judd-young/Desktop/parsec-3.0/pkgs/kernels/canneal/inputs/10.nets",
        "apparg4" : "32",

        "arielmode" : "1",
	"clock":str(clock),#add by judd
        "memmgr.memorylevels" : "2",
        "arielinterceptcalls" : "0",
        "memmgr.defaultlevel" : "0",
	"checkaddresses":1#add by judd
}

derrickgreenspan, 2020-05-17:
You need to use Intel PIN 2.14. Later versions are incompatible with SST.