sstsimulator/sst-elements #1027: Dependency Zoltan 3.83 has fatal runtime error when built with GCC 7.3 on Ubuntu 18.04

jwilso, 2018-05-10:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
The Zoltan 3.83 dependency builds under GCC 7.3 using the standard Zoltan build instructions, but at runtime there is a fatal error.

2 - Describe how to reproduce
Build SST 8.0 sufficient to run SST Scheduler Test "test_zoltan_2" in the scheduler test suite. The following error will be generated:

```
test_zoltan_2
~/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/devel/trunk/sst-elements/src/sst/elements/ember/test ~/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/devel/trunk
mpirun -np 2   /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/local/sst-core/bin/sst --verbose --partitioner zoltan --output-partition /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/devel/trunk/test/testOutputs/test_zoltan2.part --model-options --topo=torus --shape=4x4x4 --cmdLine="Init" --cmdLine="Allreduce" --cmdLine="Fini" /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/devel/trunk/sst-elements/src/sst/elements/ember/test/emberLoad.py > /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/devel/trunk/test/testOutputs/test_zoltan2.out 2>/home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/devel/trunk/test/testOutputs/test_zoltan2.err
 
WARNING: sst did not finish normally, RETVAL=255
-rwxr-xr-x 1 sstbuild sstbuild 559128 Apr 16 12:55 /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/local/sst-core/bin/sst
shunit.ASSERT: WARNING: sst did not finish normally, RetVal=255
And the Error File  (first 10 lines):
[0] Zoltan ERROR in Zoltan_Comm_Do_Post (line 180 of ../../src/Utilities/Communication/comm_do.c):  nrecvs not zero, but recv_data = NULL
[0] Zoltan ERROR in Zoltan_PHG_Build_Hypergraph (line 131 of ../../src/phg/phg_build.c):  Error getting graph from application
[0] Zoltan ERROR in Zoltan_PHG (line 291 of ../../src/phg/phg.c):  Error building hypergraph.
[0] Zoltan ERROR in Zoltan_PHG (line 504 of ../../src/phg/phg.c):  Error partitioning hypergraph.
[0] Zoltan ERROR in Zoltan_LB (line 482 of ../../src/lb/lb_balance.c):  Partitioning routine returned code -1.
FATAL: SST::Core::ZoltanPart[performPartition:273 on Rank 0] # Error using Zoltan, partition could not be formed correctly.
SST Fatal Backtrace Information:
    0 : /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/local/sst-core/bin/sst(SST::Output::fatal(unsigned int, char const*, char const*, unsigned int, char const*, ...) const+0x409) [0x558d902d4079]
    1 : /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/local/sst-core/bin/sst(SST::Partition::SSTZoltanPartition::performPartition(SST::PartitionGraph*)+0x1f4) [0x558d9030c854]
    2 : /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/local/sst-core/bin/sst(main+0x557) [0x558d9026e2e7]
    3 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f6379875b97]
    4 : /home/sstbuild/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/local/sst-core/bin/sst(_start+0x2a) [0x558d9027c25a]
~/jenkins/workspace/SST__X_ubuntu_18.04_tester/1547_2018-04-16_12-49-35/devel/trunk
```

3 - What Operating system(s) and versions 
Ubuntu 18.04 with GCC 7.3

4 - What version of external libraries (Boost, MPI)
Standard dependency set for SST v8.0

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

allevin, 2020-04-03:
@jwilso @jpvandy is this still an issue?

allevin, 2020-10-14:
Issues exist with zoltan on this platform.  We wont fix right now.  Will investigate for SST 11.

allevin, 2020-10-19:
Zoltan test is also failing on CentOS 8 using GCC 8.3.

gvoskuilen, 2022-05-17:
Zoltan was deprecated and is now removed - no longer an issue.

closed 2022-05-17 by gvoskuilen