sstsimulator/sst-elements #1054: [SST 7.2] Ariel Hang

hughes-c, 2018-06-02:
@VamseeReddyK @derrickgreenspan

CentOS 7.4 (3.10.0-693.el7.x86_64) openmpi-1.8.3, gcc-4.9.2

Ariel sends memory requests to memory hierarchy and when maximum pending transaction requests are reached Ariel waits for at least one of the pending requests response to process next request.   

- Sometimes Ariel reaches maximum pending transaction count and waits forever for any of the previous request response from memory hierarchy. Ariel is getting ticked and printing out maximum pending transaction reached forever.  
- Sometimes Ariel reaches maximum pending transaction count and memory hierarchy responds and then Ariel processes next request.

hughes-c, 2018-06-04:
@derrickgreenspan I have been able to reproduce your problem locally (the fencing issue) but not the more general problem reported by @VamseeReddyK.

derrickgreenspan, 2018-06-04:
@hughes-c  - is this on all kernels, or just LINUX-3.10.0-693.el7.x86_64?

hughes-c, 2018-06-22:
It looks like this is an issue with Samba rather than with Ariel. Closing this.

closed 2018-06-22 by hughes-c