sstsimulator/sst-elements #1055: Creating limited traces for Prospero

yarcod, 2018-06-04:
I am having some trouble starting my tests with SST and HMCSim. I orignally tried using Ariel in order to avoid trace files, but since the run time of that would occupy too much time on the department's cluster, I need something else. Thus, I would like to:
 - Create a trace file of reasonable size of just the memory operations. i.e. those that only actually arrive to the HMC(s). I have limited available disk space on aforementioned cluster, and need to keep size as small as possible.
 - To get the execution time down I would like to be able to fast-forward a number of instructions, e.g. 1 billion, and then run another billion instructions. I.e. I want to start recording after 1 billion instructions and then end recording/stop trace after another 1 billion.

I have not been find any relevant documentation on this from your website, unfortunately. It feels like something fairly basic, and if there is guide somewhere I'll happily read it before asking for more help.

Thanks!

hughes-c, 2018-06-04:
Some of the memory backends have tracing capabilities and Ariel has one that can be enabled using the tracegen flag. Ariel also has the ability to run natively until a magic instruction is encountered (ariel_enable()) as well as the ability to only run for a limited number of instructions using the max_insts flag.

With a little hacking, you should be able to get ariel to run for 1b ins segments using the magic instruction and the max_insts flag. It's also possible that we can add this capability for you but I can't give you an ETA.

yarcod, 2018-06-05:
Thanks a lot for that! I had somehow missed that there was a max_insts flag available. That should actually let me just use Ariel directly, without any need to store the trace on the limited disk space. 

What confuses me from your answer is that you indicate that some "hacking" is included, but just adding a flag to a python file (as well as changing Ariel mode) does not count for much, does it? It feels like I am missing something here. Won't setting the flag be enough from the simulator point of view?

I will, however, need to do some hacking on the SPEC benchmarks in order to add this magic instruction. This will require some investigation on my end, as I don't really know all that much about SPEC.

hughes-c, 2018-06-06:
If you want to do multiple skips, you will need a way to turn trace capture on and off. Otherwise, you will just begin when ariel_enable() is encountered and run until max_inst.

yarcod, 2018-06-08:
The latter behaviour really is all I want right now, because the other would, in my opinion, require a better knowledge about the SPEC benchmarks. What you could use instead would be something like SimPoints, which analyses the program execution and tries to give you a (shorter than original) sequense that characterizes you application. Do you have any plans to enable support for SimPoints?

Xiaoyang-Lu, 2018-06-14:
I want to know how to add the "max_insts" flag, is that means I have to add an argument "max_insts" in the python configuration for Ariel? Could you show an example on that?

hughes-c, 2018-06-14:
@Xiaoyang-Lu 

```
ariel.addParams({
    "verbose" : 1,
    "clock" : clock,
    "maxcorequeue" : 1024,
    "maxissuepercycle" : 3,
    "maxtranscore": 16,
    "pipetimeout" : 0,
    "corecount" : groups * cores_per_group,
    "memorylevels" : 1,
    "pagecount0" : num_pages,
    "pagesize0" : page_size * 1024,
    "defaultlevel" : 0,
    "arielmode" : 0, # IMPORTANT: Assumes your application has an "ariel_enable()" in it, otherwise change to 1 (no enable - start simulation immediately) or 2 (auto-detect)
    "max_insts" : 500000000,
#    "appargcount" : 4,
#    "apparg0" : "-s",
#    "apparg1": "small",
#    "apparg2" : "-t",
#    "apparg3": "8",
#    "launchparamcount": 1,
#   "launchparam0": "-ifeellucky",
    "executable" : Executable
})

```

Xiaoyang-Lu, 2018-06-14:
@hughes-c 
Thank you !

yarcod, 2018-07-01:
Thanks for the configuration example, @hughes-c! 

I do however still seem to need some help as I cannot get it working. I am trying to run ariel_enable() from within arielcore.cc after a set number of instructions (the number is set by introducing a new config flag). However, after some testing I have realized that arielcpu is never called unless profiling is already active. This means that doing a check on the instruction count from here is pointless and will never call the ariel_enable function. Where would I put this call instead?

nmhamster, 2018-08-02:
@yarcod - make sure I understand your use case, you would like to start tracing memory requests at the memory controller, mid way through execution (i.e. the trace is not being captured by the processor core (Ariel) but at the moment system)?

yarcod, 2018-08-06:
@nmhamster - yes, you are correct. I have however managed to implement something that looks to be working :)

closed 2018-08-23 by hughes-c