sstsimulator/sst-elements #1056: [8.0] Ariel Hang on fence()

hughes-c, 2018-06-04:
When the pending transaction count is 0 and there is a fence, it will hang because the call to unfence() only comes during handleEnvent(). This is unrelated to Issue https://github.com/sstsimulator/sst-elements/issues/1054.

hughes-c, 2018-06-04:
@derrickgreenspan

derrickgreenspan, 2018-06-04:
Thank you. This is the issue I am having and I am glad that it does not appear to be related to any particular kernel.

derrickgreenspan, 2018-06-05:
I need to check again, but it seems as if the issue has been fixed in the latest updates to the devel branch.

I am however getting other weird results that do not make sense. For example, the simulation time for two nodes is much smaller than for one node, even though the changes should not affect that. 

Continuing to investigate.

derrickgreenspan, 2018-06-05:
Okay, the issue where simulation time declines based on number of nodes is consistent on both my virtual machine running CentOS 7.3 and the Supercomputing cluster. I cannot reproduce @VamseeReddyK 's errors - at least, any issues I find are not related to the kernel.

hughes-c, 2018-06-05:
Closing this then.

closed 2018-06-05 by hughes-c