sstsimulator/sst-elements #1077: Is it possible to use merlin only?

Xiaoyang-Lu, 2018-07-02:
My rough plan is to simulate a system as below: Each machine has locally attached memory, with a part assigned to a pool of cluster memory. Machines are connected by a commodity network. So I want to know how to simulate the network traffic (congestion) between the different nodes, I know Merlin may has simple traffic generation models (I don't know the simple traffic generation model is enough or not)? Or just using Merlin to bridge nodes simply is enough? I cannot find any example or documents about that, so could you give me any suggestion or give a specific example about that? 
[image]
Thanks.

nmhamster, 2018-08-02:
What topology would you like to connect the memories and processors with?

Xiaoyang-Lu, 2018-08-02:
@nmhamster I use star topology now.

nmhamster, 2018-08-02:
You should be able to configure a Merlin Torus topology with a single router and `N` ports that mean everything can get routed directly.

Xiaoyang-Lu, 2018-08-02:
@nmhamster Yes. I used a single router and configured the simplest star topology, connected the nodes to this router directly.

allevin, 2020-04-03:
No further response from author, closing this issue.

closed 2020-04-03 by allevin