sstsimulator/sst-elements #1113: Prospero with ramulator

tks2004, 2018-09-18:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

Can I use ramulator for DDR4 simulation as a memory component with prospero. The trace file is generated through tracetool (sstmemtrace) of prospero. This is with SST 8.0.0

2 - Describe how to reproduce

3 - What Operating system(s) and versions 

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

hughes-c, 2018-09-18:
You can find an example of ramulator in memHierarchy/tests/sdl5-1-ramulator.py and an example of reading a binary trace in prospero/tests/array/trace-binary-withdramsim.py. You should be able to start with the prospero example and modify it to use ramulator -- essentially replace the parameters in the comp_memory component to use ramulator instead of dramsim.

tks2004, 2018-09-19:
Thanks Hughes!

closed 2018-09-19 by tks2004