sstsimulator/sst-elements #1121: access_time parameter in the Memory Controller component of memHeirarchy

tks2004, 2018-09-20:
Is there a document or description of access_time parameter for memory controller. I do not see this parameter registered for DDRSim2 and Ramulator backend in the memHeirarchy component.

hughes-c, 2018-09-20:
DRAMSim2 and Ramulator have their own configuration files. There are examples in the /tests directory: DDR3_micron_32M_8B_x4_sg125.ini and ramulator-ddr3.cfg. The parameters in each of these affect the latency for memory operations.

There are examples for other DRAMSim2 configurations in the DRAMSim2 [repo](https://github.com/umd-memsys/DRAMSim2/tree/master/ini).
There are examples for other Ramulator configurations in the Ramulator [repo](https://github.com/CMU-SAFARI/ramulator/tree/master/configs).

hughes-c, 2018-09-20:
In Prospero the trace should be bound to the cpu component that the reader is attached to.

closed 2018-09-21 by tks2004