sstsimulator/sst-elements #1151: mlm_malloc() much faster than equivalent malloc() call in Ariel

derrickgreenspan, 2018-10-23:
I've been having a rather odd issue with SST -- I notice that when I call mlm_malloc(), it is faster than if I just use malloc() normally and manually allocate memory, even if my mlm_malloc( ) and my malloc() calls are allocated to the same level.

This issue occurs regardless if I am using Opal or not. My current SST script is set up so that there are four Ariel CPUs with their own L1 and L2 caches, and a shared L3, main memory, and Non-Volatile (slower) memory, but this happens regardless of how I configure it. Even if I am not using Non-Volatile memory, and only using the L1/L2 cache and main memory, mlm_malloc() is still multiple times faster than malloc(), even though they should be using the same memory subsystem.

Does anyone have any suggestions? This is something that needs to be fixed soon for a conference deadline.

hughes-c, 2018-10-25:
@derrickgreenspan What do you mean by faster? Can you scale back to a single Ariel component and report the cycle counts for malloc() and mlm_malloc()? Also, what is your backing store model?

derrickgreenspan, 2018-10-25:
The execution time is far less with mlm_malloc() compared to malloc. For example, 3.6ms compared to 2.6ms. 

What seems to be odd is that the problem may have fixed itself after I set the statistics load level to 16... not sure why. 

The backing store model is: DDR3_micron_32M_8Bx4_sg125.ini, though I have "do_not_back" set to 1. I have the directory controller for the DRAM connected to the DRAM, and that directory controller is connected to the merlin.singlerouter.

hughes-c, 2018-10-29:
From @derrickgreenspan via email:
I think I figured out why it is happening:

mlm_malloc() doesn't seem to cause any delay and happens instantly, in contrast to malloc() which takes around .14 of a second. You can verify this for yourself by using clock(). 

ArielComponent[arielcore.cc:339:createAllocateEvent] Generated an allocate event, vAddr(map)=47283943067648, length=26001408 in level 0 from IP 2b011294f63e
ArielComponent[arielcore.cc:718:handleAllocationEvent] Handling a memory allocation event, vAddr=47283943067648, length=26001408, at level=0 with malloc ID=47283606713918
0.000000
ArielComponent[arielcore.cc:339:createAllocateEvent] Generated an allocate event, vAddr(map)=30339088, length=4 in level 0 from IP 0
ArielComponent[arielcore.cc:718:handleAllocationEvent] Handling a memory allocation event, vAddr=30339088, length=4, at level=0 with malloc ID=0
0.130000
0: Perform a mlm_malloc from Ariel 26000000, level 0
Requested: 26000000, but expanded to: 26001408 (on thread: 0) 
0: Ariel mlm_malloc call allocates data at address: 0x2b01282e3000
ArielComponent[arielcore.cc:339:createAllocateEvent] Generated an allocate event, vAddr(map)=47283969077248, length=26001408 in level 0 from IP 2b0126391702
ArielComponent[arielcore.cc:718:handleAllocationEvent] Handling a memory allocation event, vAddr=47283969077248, length=26001408, at level=0 with malloc ID=47283936237314
0.010000

See that mlm_malloc takes no more than .01 seconds, but malloc() takes .13 seconds here.

Is there something that needs to be changed in order to get mlm_malloc() to take time? Otherwise, my best bet would be to just compare mlm_malloc(0) to mlm_malloc(1) when doing tests, and change every reference to malloc() in my code to mlm_malloc().

gvoskuilen, 2018-10-29:
@derrickgreenspan Are you concerned about the two events taking a different amount of real/system time (measured in Ariel with clock()) or are they actually taking a different amount of simulated time (measured with Ariel's 'currentCycles' variable)? mlm_malloc and malloc are different function calls so they will take a different amount of real time, but they should take the same amount of simulated time since both are handled as ARIEL_ISSUE_TLM_MAP commands by the ariel cores. Enabling statistics also causes real/system time to increase which is why you see a difference when enabling those.

derrickgreenspan, 2018-10-29:
They are in fact taking a different amount of simulated time and having different cycle counts, which is the confusion:

For example, the cycle count for a simple malloc was: 459399 cycles, while for an mlm_malloc it was: 333922 cycles. This causes the simulated time to be different too: 166.961us vs 229.7us, and this is for a simple program that just allocates and then iterates an array of integers and then returns!

For large programs, the simulated time can be easily an order of magnitude different.

One thing I do notice is that when I use malloc(), it doesn't seem like slurm recognizes it; for example, I don't get a notification that ARIEL_ISSUE_TLM_MAP was ever issued at all, although perhaps I need to increase the verbosity of my script (or there's something else I need to specify)

Thanks for your help. My current solution is to just replace any malloc() with mlm_malloc referring to the default level.

gvoskuilen, 2018-10-29:
Ok, in that case, can you describe your measurement process for the cycle counts? Are you counting from the application, from fesimple, or from Ariel itself? Where do you start counting and where do you end? What functions/variables are you using to measure time? 300K-400K cycles is very high for either of these calls.

What do you mean by 'slurm' recognizing it? If you mean that the pintool isn't generating any output when it finds a malloc, that's correct. There's a commented out printf in fesimple that you could uncomment if you'd like. Your output above shows that Ariel is getting the request in both cases ("Generated an allocate event...") so that part looks right.

derrickgreenspan, 2018-10-30:
Okay, sorry for not providing enough detail;

My program simply iterates across an array 10000 elements large, setting the value of each element in the array to a particular integer. I am counting from ariel itself, and measuring the cycle count from the ariel_enable call to the end of the function. Specifically, I am looking at the variable: a0.cycles.0.

In retrospect, the bigger issue really isn't the time/cycle count that mlm_malloc() or malloc() takes; the issue is that operations on that allocated block of memory take less time if called via mlm_malloc() vs. malloc(). I'm not sure how to fix that.

gvoskuilen, 2018-11-01:
Ah. That makes sense now, thanks for the explanation. I think the issue then is down to the different implementations, and your test being short enough that a malloc call constitutes a large fraction of the instructions executed. Ariel executes malloc by simulating the actual malloc call, and then intercepting its return to issue the MAP instruction. In contrast, mlm_malloc is simulated by replacing the mlm_malloc call with a call in the pintool that internally calls memalign and issues a MAP instruction. As such, in malloc, the syscall is simualted and in mlm_malloc, it is not. In a short test like this, the difference will be apparent.  

Your solution is a good workaround. If you do want to capture allocations that aren't explicitly malloc (new, vector.insert(), etc.), or you want to simulate the malloc call, an alternative is to place a call to "ariel_malloc_flag()" before a call to malloc/new that should go to a non-default level/memory pool. I can explain that further if you want to go that route.

derrickgreenspan, 2018-11-18:
Hi, I do have a question -- if I am doing a memory allocation that is static (like... int test[10000] = {3}), is there a way to specify if which memory level it should go to?

hughes-c, 2018-11-19:
```
ariel_mmap_mlm(int fileID, size_t size, int level)
    ariel_mmap_mlm(42, sizeof(uint32_t) * NUM_INTS, 2)
```
  
```
ariel_mlm_malloc(size_t size, int level)
    ariel_mlm_malloc(sizeof(uint32_t) * NUM_INTS, 2)
```