sstsimulator/sst-elements #1165: How to use SST arguments inside Ember motif

Vixz7, 2018-11-12:
Hi,
I am trying to run an Ember Motif from "ember/mpi/motifs" with following command-line:

> sst --model-options="--cmdLine=Init --cmdLine=\"Halo3D26 iterations=10 \" --cmdLine=Fini --numNodes=27 --numCores=1 --topo=fattree --shape=1,1:27 --netPktSize=4096B --netBW=0.7GB/s" $SST/sst-elements-library-8.0.0/src/sst/elements/ember/test/emberLoad.py

I am exploring SST and trying to understand the way it works. I tried multiple ways to call the platform parameters (command-line arguments) inside motif source code, without success.

How can I use the hardware specification like "numNodes" and "numCores" inside the motif ?
or in other way
How can I call these command-line arguments inside motif source ?

Any helps or insights are highly appreciated.

Thanks

nmhamster, 2019-01-22:
@Vixz7 - good question. We don't export these to the motifs today. How would you like this to work - a function to query something in the environment? Right now, a slightly cheat way to do this would be to export an environment variable (e.g. `MY_MOTIF_PARAMS=1000` and then to call `getenv("MY_MOTIF_PARAMS")` inside the motif. Nasty though. Let us know on the interface you would like. Thanks for the question!

Vixz7, 2019-01-29:
Thank you @nmhamster for your response.
Currently, I am using a temporary variable in Motif (RPN) and passing on the value of environment variable (numCores) to this temporary variable inside the run configuration file (~/sst-elements-library-8.0.0/src/sst/elements/ember/test/emberLoad.py).
I did below changes in the loadInfo.py ( called inside emberLoad.py) to achieve this:
--------loadInfo.py--------------
```
< if 1 == cmdNum:
<       tmp     = 'motif' + str(cmdNum) + '.arg.' + 'RPN'
<       motif[ tmp ] = `self.numCores`
```
----------------------------------
Above arrangement solves my purpose as of now.
I think using a function inside motif to query environment variable will be a better approach here.

Thank you,
~V

allevin, 2020-04-03:
No further response from author, closing this issue.

closed 2020-04-03 by allevin