sstsimulator/sst-elements #1169: Memory subsystem with Merlin

tks2004, 2018-11-13:
Iam trying to emulate the scenario of running a trace file from one numa domain and the memory is allocated on the second numa domain. To simulate this I have a model using "prospero" CPU model with shared L3 connected to merlin router0 that has directory controller connected to this router. The memory controller is connected to the another merlin router 1.  Router 0 and router1 are linked with "ports".

L3 ( "directory" link) -> router0 ("port1")
Directory Controller ("network") -> router0("port2")
memory controller ("network" link") -> router1 ("port1")
router0 ("port3" link) -> router1("port2") 

I see the following error:
Thread 1 "sstsim.x" received signal SIGSEGV, Segmentation fault.
SST::Units::operator== (this=this@entry=0x522d480, lhs=...) at unitAlgebra.cc:278
278         if ( numerator.size() != lhs.numerator.size() ) return false;

I am using the source from mh_ports branch. Is this scenario supported.

tks2004, 2018-11-20:
Hi, Could you please help me with the above error

hughes-c, 2018-11-20:
Can you post your python configuration?

nmhamster, 2018-11-20:
You might also want to run SST with `--dump-config <filename here>` because this will generate a Python file, post graph generation. Uploading this file to Github would be useful. This error is likely to occur because one of the parameters is not correctly formed to generate a Unit Algebra.

tks2004, 2018-11-27:
Hi, From the configuration and post configuration files, the links bewteen components looks to be OK. Could you please check and let me know, If Iam missing anything here.

gvoskuilen, 2018-11-27:
Do you have anything on port0 on the routers? The port numbering should start from 0 not 1.

tks2004, 2018-12-04:
After port numbering were corrected, the directory controller sends the request to MC on the second router, but there is no retrurn from the MC. The simulation hangs

40 (dirctrln0) Processing: ID: <16,0> Cmd: GetX Src: l3n0 Dst: dirctrln0 Rqstr: cpu0l1n0 Flags: [] MemFlags: 0x0 Addr: 0x1001 BaseAddr: 0x1000 (Global) VA: 0x1 IP: 0x0 Size: 64 Prefetch: false
        MSHR: Event Inserted. Key addr = 1000, event Addr = 1001, Cmd = GetX, MSHR Size = 1, Entry Size = 1
        Requesting data from memory.  Cmd = GetX, BaseAddr = x1000, Size = 64, noncacheable = false

41 (dirctrln0) Sending event to memory: ID: <16,0> Cmd: GetX Src: dirctrln0 Dst: memoryn0 Addr: 0x1001 BaseAddr: 0x1000 Size: 64

Attached the Post processed and the file being used during simulation. Could you please check if Iam missing anything here.
[post_sst.py.txt](https://github.com/sstsimulator/sst-elements/files/2644455/post_sst.py.txt)
[1c_prop_try3.py.txt](https://github.com/sstsimulator/sst-elements/files/2644456/1c_prop_try3.py.txt)

tks2004, 2018-12-06:
Could you please check if Iam missing anything in the python script attached in the previous thread.

allevin, 2020-04-03:
No further response from author, closing this issue.

closed 2020-04-03 by allevin