sstsimulator/sst-elements #1192: Installation of Hermes and other headers

jjwilke, 2018-12-19:
Most element libraries specify:

````
libX_la_SOURCES = \
  myFile.h \
  myFile.cc
````

rather than
````
libX_la_SOURCES = \
  myFile.cc

nobase_library_include_HEADERS = \
  myFile.h
````

Headers do not get installed. There's obviously a need to avoid circular dependencies, but in my case (putting macro on top of Hermes) I would prefer to point macro at installed headers rather than a source tree.

Is there a reason I should avoid installed headers as needed? Particularly hermes - which does not appear to have any actual source files?

nmhamster, 2019-01-31:
@jjwilke - can you see if #1203 fixes this issue for you?

nmhamster, 2019-02-05:
@jjwilke - closed, please re-open if needed.

closed 2019-02-05 by nmhamster