sstsimulator/sst-elements #1225: Run 433.milc (SPEC2006), with input data being piped from file

yarcod, 2019-02-25:
I want to run a couple of SPEC benchmarks through the simulator, but I am having issues starting a few of them. One being 433.milc, and the problem is that this specific benchmark application takes its input from stdin, i.e. milc < input.in. I am not able to perform this pipe from within the Python configuration file -- how would I go about doing that?

While the above probably has a relatively simple solution, I also am having trouble launching 429.mcf. However, I am suspecting that this is due to something that I have added to the simulator. I just wanted to ask if anyone else have had any similar issues with mcf. 

Thanks in advance,

nmhamster, 2019-02-25:
@yarcod - hmmm.. good question. We will need to look at that. What errors are you getting on the `mcf` run?

nmhamster, 2019-02-26:
@yarcod - we think we might have a way to modify Ariel to get you stdin from a file but it will need some (minor) code changes. Are you OK using SST from the devel branch in our repo?

yarcod, 2019-02-26:
@nmhamster I am having some really weird issues with running mcf on a compute cluster. When trying to simulate the application on the _login node_, it works without a hitch. However, submitting the job to the cluster does not appear to start the simulation correctly. I have tried to analyse the reason for this, but to no avail. I can see that the clock is ticking, but it seems like the simulator is not starting fully. And I have not seen this behaviour when running any other application. I'm just dumbfounded as to why it would be behaving like this.

I would prefer to keep the version I using right now (7.2.0), but I would have no problem doing a cherry-pick on the specific commit that would fix the issue. Could that work?

nmhamster, 2019-03-01:
@yarcod - can you try enabling the SST heartbeat (`--heartbeat-period=PERIOD`), set `PERIOD` to something sensible for your simulation e.g. `1ns` for on-node simulations. This should print out a message periodically from the SST core so you can at least see the internal simulation time is moving forward. If there is a serious problem, the messages will not print correctly.

Cherry picking is fine. We think we have a technical solution to your issue but we need to add some testing here.

yarcod, 2019-03-06:
I... am speechless. For some utterly unexplainable reason, enabling the heartbeat actually made the simulation progress! Despite so many attempts to get things running I have until now never seen mcf actually print anything what so ever. Well, no matter, it seems to be working.

Happily awaiting a way to run milc as well! :)

yarcod, 2019-03-14:
Just wanted to check on the status on this issue. If you have a solution that you think works fine, I'm okay with taking that and do some manual testing on my end. I'd love to be able to start running Milc soon :)

allevin, 2020-04-03:
@yarcod what is the status of this issue?  If no longer a problem, we would like to close it.

yarcod, 2020-04-13:
@allevin well, as far as I could tell, there never came any solution to this problem. However, I have not checked since I wrote my last inquiry and just decided not to run milc in the end.

allevin, 2020-04-13:
@yarcod Thank you, I am going to close this issue.  If you have issues again, open a new issue.

closed 2020-04-13 by allevin