sstsimulator/sst-elements #1231: Should I use STT-Macro or STT-Merlin

KongXC123, 2019-03-06:
Hi, All
    I want to evaluate the performance of  HPC Network(may be DragonFly or Fat-tree), and the network has some features such as dynamic routing and congestion control, which I want to evaluate the effect with HPC benchmark. 
    I think I can modeling the Network with either STT-Macro or STT-Merlin, but I  want to know which tool is suiltable for me. 
    Any suggestion should be great. 
    Thank you.

nmhamster, 2019-03-06:
@KongXC123 - it depends on what kind of models you want to utilize. For instance, do you want to use traces, communication motifs or do you want to generate a skeleton of your application manually? 

- SST-Macro and SST-Micro both provide tracing capability
- SST-Micro provides predefined communication patterns for basic HPC benchmarks (see the SST-Ember component)
- SST-Macro and SST-Micro both provide skeleton-like capabilities but the SST-Macro implementation may be easier to use.

In general both tools can be used to model adaptive routing with DragonFly and Fat-Tree networks. Congestion control is an emerging area in the models so you may want to update frequently.

Does that help? We are happy to talk more to help.

KongXC123, 2019-03-07:
@nmhamster 
Hi, 
    Thank you very much. It does help, and I think SST-Micro is more suitable for me because of  "predefined communication patterns for basic HPC benchmarks".
    Best regards.

nmhamster, 2019-03-08:
@KongXC123 - you may want to look at the patterns found here: https://github.com/sstsimulator/sst-elements/tree/master/src/sst/elements/ember/mpi/motifs these are useful for modeling MPI communication.

nmhamster, 2019-03-08:
I will close this issue, please let us know (reopen the issue) if you have any additional questions or feedback.

closed 2019-03-08 by nmhamster