sstsimulator/sst-elements #1255: Missing return warnings for GCC for switch statements

jjwilke, 2019-04-08:
Several switch statements do not provide a trailing return or default case statement, leading GCC to spew complaints about missing return statements in memOp.h, nicEvents.h, and shmemapi.h

hughes-c, 2019-07-16:
@jjwilke Should this go into 9.1?

jwilso, 2019-10-21:
@jjwilke @hughes-c Was this ever addressed? I will leave it open if so.

hughes-c, 2019-10-22:
@jwilso Not that I'm aware of.

nmhamster, 2021-02-19:
@allevin / @jjwilke - do we have a list of these we can knock out for the release?

allevin, 2021-02-19:
@jjwilke do you know what version of the gcc compiler was throwing the complaints?   
I will try to do a warnings scrub next week.

allevin, 2021-04-07:
@jjwilke @nmhamster I cannot reproduce these build warnings.  I tried GCC 4.8 and GCC 6.1... The issue was submitted in 2019, and may have been cleaned up since.  I recommend closing.

closed 2023-07-03 by hughes-c