sstsimulator/sst-elements #1286: How to change Ember Motif parameters using emberLoad.py

JaGualda, 2019-06-03:
New Issue for sst-elements
--------------------------

**> 1 - Detailed description of problem or enhancement**
I'm trying to change the grid size of the Halo3D motif using the emberLoad.py file, but I can't find any resource on how to do it. 

**2 - Describe how to reproduce**
This doesn't seem to work.

> sst --model-options="--cmdLine=Init --cmdLine="Halo3D nx=128 ny=128 nz=128" --cmdLine=Fini --topo=dragonfly --numCores=4 --shape=64:2:2:4" emberLoad.py

`

**3 - What Operating system(s) and versions** 
Ubuntu 18.04.2

nmhamster, 2019-06-05:
@jaGualda - you can set up the default simulation workflow (for motifs) in `defaultSim.py`

```
def getWorkFlow( defaults ):
    workFlow = []
    motif = dict.copy( defaults )
    motif['cmd'] = "Init"
    workFlow.append( motif )

    motif = dict.copy( defaults )
    motif['cmd'] = "Sweep3D nx=30 ny=30 nz=30 computetime=140 pex=4 pey=16 pez=0 kba=10"     
    workFlow.append( motif )

    motif = dict.copy( defaults )
    motif['cmd'] = "Fini"
    workFlow.append( motif )

	# numNodes = 0 implies use all nodes on network
    numNodes = 0 
    numCores = 1 

return workFlow, numNodes, numCores 
```

nmhamster, 2019-06-05:
@mjleven - should we recommend any thing else?

JaGualda, 2019-06-05:
Right. I thought about that but I was wondering if there was a way of doing it from  the cmdLine. That will do for now.

Thanks.

closed 2019-06-05 by JaGualda