sstsimulator/sst-elements #1296: Compile ariel even when Pin is not available

calewis, 2019-06-19:
It would be helpful if ariel can be compiled with a normal configure even when pin is not available, if there are no other hold ups then this would more easily allow for mac development as well.

derrickgreenspan, 2019-06-26:
I believe this would not be very useful, since without Intel Pin, dynamic analysis of executing programs would not be possible anymore. 

That being said, the HPCC that I work with will soon be deprecating support for older gcc versions (gcc 4.9.2 and below,) which means we won't be able to support Intel PIN for much longer. If, somehow, we are able to get Intel PIN 2.14 (and thus, ariel) working on newer versions of GCC (something I've been trying to do), it'll soon be broken once we upgrade to centos 8, which uses the 4.18 kernel. We anticipate this upgrade happening in December 2019.

calewis, 2019-06-26:
@derrickgreenspan, the idea would be to work on a pin replacement, thus wanting configure to compile it even without pin.

derrickgreenspan, 2019-06-26:
@calewis -- Okay, that makes sense. I thought you wanted to be able to compile it even without a PIN replacement.

gvoskuilen, 2020-10-21:
Done

closed 2020-10-21 by gvoskuilen