sstsimulator/sst-elements #1305: SST Stake fails to build on master with changes to generator API

jleidel, 2019-07-01:
The latest master branch of sst-elements fails to build with the SST Stake Miranda generator enabled.  This bug is induced by commits from 7c4711da37951ffb7508bf350ed5f16ac92c1444

I have a functional fix and will submit a separate pull request to the appropriate devel branch

Trace: 
generators/stake.cc:25:1: error: prototype for ‘SST::Miranda::Stake::Stake(SST::ComponentId_t*, SST::Params&)’ does not match any in class ‘SST::Miranda::Stake’
 Stake::Stake( ComponentId_t* id, Params& params ) :
 ^~~~~
In file included from generators/stake.cc:19:0:
../../../../src/sst/elements/miranda/generators/stake.h:42:7: error: candidates are: SST::Miranda::Stake::Stake(const SST::Miranda::Stake&)
 class Stake : public RequestGenerator {
       ^~~~~
../../../../src/sst/elements/miranda/generators/stake.h:46:2: error:                 SST::Miranda::Stake::Stake(SST::ComponentId_t, SST::Params&)
  Stake( ComponentId_t id, Params& params );
  ^~~~~
../../../../src/sst/elements/miranda/generators/stake.h:45:2: error:                 SST::Miranda::Stake::Stake(SST::Component*, SST::Params&)
  Stake( Component* owner, Params& params );

jleidel, 2019-07-01:
@nmhamster @gvoskuilen feel free to assign this bug to me directly

gvoskuilen, 2019-07-01:
Thanks @jleidel, I missed that one. We've been updating of the subcomponent API. The new subcomponent constructor is "SubComponent(ComponentId_t id, Params& params) but for now the old constructor with the component pointer is still required.

jleidel, 2019-07-01:
No worries!  I should have a pull request ready in a few minutes

jleidel, 2019-07-02:
Merged in pull request #1308; Closing

closed 2019-07-02 by jleidel