sstsimulator/sst-elements #1338: How to get stats when using Ramulator backend for MemController?

shafeey, 2019-07-30:
Hi,

I wanted to use Ramulator for memory backend. When I use DRAMSim2, the DRAMSim2 stats are saved to 'dramsim.log' file. Is it possible to enable similar stats for Ramulator? Normally Ramulator will create a file DDR3.log for statistics when run in trace mode.

I have looked into the backend parameters. Only 'configFile' and 'verbose' are available. But I can't see 'verbose' printing anything to 'sstoutput' file.

I am using the latest master branch for sst core and elements. Can someone tell me if it's possible to do what I want?

Thank you.

hughes-c, 2019-08-05:
The memory stats are dumped into an SST statistics object. You can look at the statistics by finding the name that you assigned the component in the python configuration. For example:

`memory = memctrl.setSubComponent("backend", "memHierarchy.ramulator")`

 ```
memory.requests_received_GetS : Accumulator : Sum.u64 = 61; SumSQ.u64 = 61; Count.u64 = 61; Min.u64 = 1; Max.u64 = 1; 
 memory.requests_received_GetSX : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 0; Min.u64 = 0; Max.u64 = 0; 
 memory.requests_received_GetX : Accumulator : Sum.u64 = 3; SumSQ.u64 = 3; Count.u64 = 3; Min.u64 = 1; Max.u64 = 1; 
 memory.requests_received_PutM : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 0; Min.u64 = 0; Max.u64 = 0; 
 memory.outstanding_requests : Accumulator : Sum.u64 = 2037; SumSQ.u64 = 2349; Count.u64 = 104054; Min.u64 = 0; Max.u64 = 2; 
 memory.latency_GetS : Accumulator : Sum.u64 = 1950; SumSQ.u64 = 63466; Count.u64 = 61; Min.u64 = 18; Max.u64 = 45; 
 memory.latency_GetSX : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 0; Min.u64 = 0; Max.u64 = 0; 
 memory.latency_GetX : Accumulator : Sum.u64 = 87; SumSQ.u64 = 2619; Count.u64 = 3; Min.u64 = 21; Max.u64 = 33; 
 memory.latency_PutM : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 0; Min.u64 = 0; Max.u64 = 0; 
 memory.cycles_with_issue : Accumulator : Sum.u64 = 64; SumSQ.u64 = 64; Count.u64 = 64; Min.u64 = 1; Max.u64 = 1; 
 memory.cycles_attempted_issue_but_rejected : Accumulator : Sum.u64 = 0; SumSQ.u64 = 0; Count.u64 = 0; Min.u64 = 0; Max.u64 = 0; 
 memory.total_cycles : Accumulator : Sum.u64 = 104054; SumSQ.u64 = 10827234916; Count.u64 = 1; Min.u64 = 104054; Max.u64 = 104054; 
```

gvoskuilen, 2019-08-05:
To add to that, there is not currently a way to enable ramulator's statistics via SST. We do have some statistics at the memory controller as @hughes-c pointed at. I will turn this thread into an Enhancement request however and we'll look at enabling this in a future release.

divija95, 2022-10-23:
Is this feature supported as of now?

gvoskuilen, 2022-10-24:
@divija95 No this is not supported.