sstsimulator/sst-elements #1348: sst-elements configures successfully despite empty sst-core prefix

jjwilke, 2019-08-19:
sst-core checks can fail (in the current case, missing libmpi) resulting in an empty core prefix. Configure still succeeds and generates makefiles. Make then crashes with strange missing header errors because sst-core is not a valid path.

hughes-c, 2019-08-19:
It looks like this can happen if you had a previously good build of core and still have some detritus lying around -- so, not a completely empty install target. The configure for elements tests for sst-config and sst-register in the target of the bin directory of the --with-sst-core option.

nmhamster, 2019-08-19:
@jjwilke - can you explain what missing libmpi means - is this SST core was compiled with MPI, but during configure of elements it is not available (e.g. `LD_LIBRARY_PATH` is changed?). We will need some additional information here because as @hughes-c says, we don't really understand how you could get into this situation.

jjwilke, 2019-08-19:
I believe in this case the problem was that core was built without MPI but elements was expecting MPI. We'll try to get a minimal reproducer config in here when we have time. I just wanted to make a note for now.

jjwilke, 2019-08-19:
I will also self-assign. This isn't intended to be anything you need to worry about.

nmhamster, 2019-08-19:
@jjwilke - I don't think we have any elements which depend on MPI being present. We don't build with `MPICC` or `MPICXX` by design to avoid that happening. I wouldn't say its impossible though, I just thought we tried to prevent accidental use as a design philosophy of SST was no direct MPI calls in elements (all must go through core so we can use threads/MPI splits wherever we want, or I guess some other alternative model).

nmhamster, 2019-08-19:
@jjwilke - OK, let us know if you need anything from us. :-)

hughes-c, 2023-07-03:
@feldergast @gvoskuilen  Is this still a thing?

berquist, 2023-07-17:
I can get partway there by not specifying `--with-sst-core`, but it still fails for a relatively clear reason:
```
configure: --with-sst-core check = good 1
configure: --with-sst-core check = good 2
checking for sst-config... no
checking for sst-config tool available... configure: error: Unable to find sst-config in $PATH
```
Putting `sst-config` on my `PATH` via the install `bin/` works ok without passing `--with-sst-core`, which again seems fine based on `config/sst_core_check_install.m4`.

This is probably not worth investigating further until someone can show a similar problem again.