sstsimulator/sst-elements #1357: Unable to find matching request

mazhulin, 2019-08-30:
when I followed SST_Tutorial_overview and learned the example(Miranda CPU<->L1 Cache<->Memory), I wrote a configuration file according to the tutor. However, when i used command (sst -v configFile.py) to run the file, there are some errors as following:

FATAL: (core interface) Unable to find matching request. Event: ID: (48,0), Cmd: FetchInv Src: L1 Dst: core Rqstr: L1 Flags:[]  MemFlags: 0x0 Addr:0x0  Baseaddr: 0x0 (Global)  VA: 0X0 IP:0X0 size:64 prefetch: false.

what should i do to resolve this problems?

hughes-c, 2019-09-03:
@mazhulin will you post the python script that you're using (configFile.py)?

mazhulin, 2019-09-04:
My configFile.py  is as following:
# Automatically generated SST Python input
import sst

# Define SST core options
sst.setProgramOption("timebase", "1 ps")
sst.setProgramOption("stopAtCycle", "100ms")

core=sst.Component("core","miranda.BaseCPU")
cache=sst.Component("L1","memHierarchy.Cache")
memory=sst.Component("memory","memHierarchy.MemController")

core.addParams({
    "clock":"2.4GHz",
    "generator":"miranda.STREAMBenchGenerator",
    "generator.n":"1000",
})
cache.addParams({
    "cache_frequency":"2.4GHz",
    "cache_size":"2KB",
    "associativity":4,
    "access_latency_cycles":2,
    "mshr_latency_cycles":2,
})
memory.addParams({
    "backend.mem_size":"20MB",
    "clock":"2.4GHz",
})

core_cache=sst.Link("core_to_cache")
cache_mem=sst.Link("cache_to_memory")

core_cache.connect(
    (core,"cache_link","100ps"),
    (cache,"high_network_0","100ps")
)

cache_mem.connect(
    (cache,"low_network_0","100ps"),
    (memory,"direct_link","100ps")
)

When i run  this config file with "sst configFile.py", it will prompt me with the following error message:
FATAL: (core interface) Unable to find matching request. Event: ID: <48,0> Cmd: FetchInv Src: L1 Dst: core Rqstr: L1 Flags: [] MemFlags: 0x0 Addr: 0x0 BaseAddr: 0x0 (Global) VA: 0x0 IP: 0x0 Size: 64 Prefetch: false
FATAL: (core interface) Unable to find matching request. Event: ID: <48,0> Cmd: FetchInv Src: L1 Dst: core Rqstr: L1 Flags: [] MemFlags: 0x0 Addr: 0x0 BaseAddr: 0x0 (Global) VA: 0x0 IP: 0x0 Size: 64 Prefetch: false
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_jS2_z+0x40a) [0x4b2bca]
    1 : /home/mzl/local/sstelements-8.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy21MemHierarchyInterface15processIncomingEPNS0_12MemEventBaseE+0xbe) [0x7f8461026cee]
    2 : /home/mzl/local/sstelements-8.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy21MemHierarchyInterface14handleIncomingEPNS_5EventE+0x11) [0x7f8461026e01]
    3 : sst(_ZN3SST10Simulation3runEv+0x1c7) [0x4b96c7]
    4 : sst() [0x460b32]
    5 : sst(main+0x1795) [0x4548d5]
    6 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f846c724830]
    7 : sst(_start+0x29) [0x4603b9]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode -1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on