sstsimulator/sst-elements #1360: Migrating Network Simulations from SST 8 to SST 9

deanchester, 2019-09-05:
I am trying to update my simulations to run in SST 9 but I am running in to an issue where it fails to initialise a linkControl correctly. I am now using the new ember load scripts. This is from the master branch commit fe4186d487bf05bd1db871a231d4b8b37cd57e54; core is at commit sstsimulator/sst-core@9515cf8a540abd8edc380e9f8e91c202d49c17e6

With the updated scripts I now run my simulations like so: 
`$ sst --model-options="--platParams=astra --jobFile=pingPong --topo=fattree --shape=2,1:2" emberLoadFile.py`

The output from the run looks like:

> Platform: configuration "astra"
> Network: topo=fattree shape=2,1:2 numNodes=4
> JobInfo: jobId=1 numNodes=2 numRanksPerNode=1
> JobInfo: nidList="0,3"
> JobInfo:    cmdLine: "Init"
> JobInfo:    cmdLine: "PingPong messageSize=4194304 iterations=10"
> JobInfo:    cmdLine: "Fini"
> Assertion failed: (m_linkControl), function Nic, file nic.cc, line 138.
> [Deans-MacBook-Pro-17:88271] *** Process received signal ***
> [Deans-MacBook-Pro-17:88271] Signal: Abort trap: 6 (6)
> [Deans-MacBook-Pro-17:88271] Signal code:  (0)
> [Deans-MacBook-Pro-17:88271] [ 0] 0   libsystem_platform.dylib            0x00007fff5f0e8f5a _sigtramp + 26
> [Deans-MacBook-Pro-17:88271] [ 1] 0   ???                                 0x0000000000002d08 0x0 + 11528
> [Deans-MacBook-Pro-17:88271] [ 2] 0   libsystem_c.dylib                   0x00007fff5ee861ae abort + 127
> [Deans-MacBook-Pro-17:88271] [ 3] 0   libsystem_c.dylib                   0x00007fff5ee4e1ac basename_r + 0
> [Deans-MacBook-Pro-17:88271] [ 4] 0   libfirefly.so                       0x000000010ee5909d _ZN3SST7Firefly3NicC2EyRNS_6ParamsE + 16349
> [Deans-MacBook-Pro-17:88271] [ 5] 0   libfirefly.so                       0x000000010edc790b _ZN3SST3ELI14DerivedBuilderINS_7Firefly3NicENS_9ComponentEJyRNS_6ParamsEEE6createEyS6_ + 43
> [Deans-MacBook-Pro-17:88271] [ 6] 0   sstsim.x                            0x000000010d4c7db7 _ZN3SST7Factory15CreateComponentEyRNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEERNS_6ParamsE + 439
> [Deans-MacBook-Pro-17:88271] [ 7] 0   sstsim.x                            0x000000010d4de5ea _ZN3SST10Simulation13performWireUpERNS_11ConfigGraphERKNS_8RankInfoEy + 1562
> [Deans-MacBook-Pro-17:88271] [ 8] 0   sstsim.x                            0x000000010d493117 _ZL16start_simulationjR15SimThreadInfo_tRN3SST4Core10ThreadSafe7BarrierE + 295
> [Deans-MacBook-Pro-17:88271] [ 9] 0   sstsim.x                            0x000000010d490062 main + 8722
> [Deans-MacBook-Pro-17:88271] [10] 0   libdyld.dylib                       0x00007fff5edda015 start + 1
> [Deans-MacBook-Pro-17:88271] *** End of error message ***
> [1]    88271 abort      sst  emberLoadFile.py --output-config=configuration.py

Here is the configuration that is being run: 

```
# Automatically generated by SST
import sst

# Define SST Program Options:
sst.setProgramOption("timebase", "1 ps")
sst.setProgramOption("stopAtCycle", "0 ns")

# Define the SST Components:
comp_nic0 = sst.Component("nic0", "firefly.nic")
comp_nic0.addParams({
     "input_buf_size" : "14KB",
     "link_bw" : "100Gb/s",
     "module" : "merlin.linkcontrol",
     "nic2host_lat" : "20ns",
     "nid" : "0",
     "num_vNics" : "1",
     "output_buf_size" : "14KB",
     "packetSize" : "2048B",
     "rxMatchDelay_ns" : "75",
     "txDelay_ns" : "50"
})
comp_nic0.setCoordinates(0, 0, 0)
link_id_00000000 = sst.Link("link_id_00000000")
comp_nic0.addLink(link_id_00000000, "core0", "20 ns")
link_id_00000002 = sst.Link("link_id_00000002")
comp_nic0.addLink(link_id_00000002, "rtr", "120 ns")
comp_loopBack0 = sst.Component("loopBack0", "firefly.loopBack")
comp_loopBack0.addParams({
     "numCores" : "1"
})
comp_loopBack0.setCoordinates(0, 0, 0)
link_id_00000001 = sst.Link("link_id_00000001")
comp_loopBack0.addLink(link_id_00000001, "core0", "1 ns")
comp_nic0core0_EmberEP = sst.Component("nic0core0_EmberEP", "ember.EmberEngine")
comp_nic0core0_EmberEP.addParams({
[... 342 lines cut ...]
comp_rtr_l1_g0_r0.addParams({
     "debug" : "0",
     "fattree:shape" : "2,1:2",
     "flit_size" : "8B",
     "id" : "2",
     "input_buf_size" : "14KB",
     "input_latency" : "50ns",
     "link_bw" : "100Gb/s",
     "link_lat" : "120ns",
     "num_peers" : "4",
     "num_ports" : "2",
     "output_buf_size" : "14KB",
     "output_latency" : "50ns",
     "topology" : "merlin.fattree",
     "xbar_bw" : "100Gb/s"
})
comp_rtr_l1_g0_r0.setCoordinates(0, 0, 0)
comp_rtr_l1_g0_r0.addLink(link_id_00000006, "port0", "120 ns")
comp_rtr_l1_g0_r0.addLink(link_id_0000000d, "port1", "120 ns")
# Define SST Statistics Options:
sst.setStatisticOutput("sst.statOutputConsole")
# End of generated output.
```

Any ideas?

feldergast, 2019-09-09:
I'm going to need your "pingPong" job file to see what's going on.  Unfortunately, with some of the more recent changes, the configuration is complex enough that the autogenerated python output isn't capturing everything (that's another issue we'll need to look into).  We can just use the default platform file, so won't need the astra platform file you appear to be using.

feldergast, 2019-09-09:
I understand from @nmhamster that this has been resolved.  Closing for now.  If you have ongoing issues, please reopen.

closed 2019-09-09 by feldergast