sstsimulator/sst-elements #1396: how to config SPEC2006 in SST to run benchmarks?

juddyoung, 2019-11-17:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

2 - Describe how to reproduce

3 - What Operating system(s) and versions 

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

gvoskuilen, 2019-11-22:
Are you trying to run this on a specific SST CPU model? Do you have a config file? If not, the simplest method would be to create a python config file for Ariel (or grab one from src/sst/elements/ariel/frontend/simple/examples/stream and point Ariel's 'executable' argument to whatever benchmark you want to run. Use Ariel's 'appargcount' and 'appargX' arguments if you need to give the benchmark input arguments.

juddyoung, 2019-11-25:
@gvoskuilen thank you so much, I know how to config now but I met another problem, after running the python configuration file there would be segmentation faults, and I want to config with multi-cores, can you show me how to solve it?
[bzip2.txt](https://github.com/sstsimulator/sst-elements/files/3886785/bzip2.txt)

jjwilke, 2019-11-25:
@juddyoung Is the simulation running and segfaulting at the end? Can you post the output of the backtrace here? Or do you possibly have a gdb backtrace with line numbers?

juddyoung, 2019-11-26:
@jjwilke it's running and the segfault does happen at the end.
this is the output after executing "sst bzip2.py"
[sst_output.txt](https://github.com/sstsimulator/sst-elements/files/3889671/sst_output.txt)
this is the output after "gdb sst" and r "home/juddyoung/mzl/bzip2.py"
[gdb_output.txt](https://github.com/sstsimulator/sst-elements/files/3889681/gdb_output.txt)

jjwilke, 2019-11-26:
Can you cut and paste the output directly into the issue tracker? It helps with searching for related issues.

juddyoung, 2019-11-26:
@jjwilke ok
this is sst output:
juddyoung@lw:~/mzl$ sst bzip2.py 
Creating Core 0 in Group 0
Creating Core 1 in Group 0
Creating Core 2 in Group 0
Creating Core 3 in Group 0
Creating Core 4 in Group 0
Creating Core 5 in Group 0
Creating Core 6 in Group 0
Creating Core 7 in Group 0
Initialized with 8 cores
Before initialization 
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
After initialization 
[lw:03495] *** Process received signal ***
[lw:03495] Signal: Segmentation fault (11)
[lw:03495] Signal code: Address not mapped (1)
[lw:03495] Failing at address: 0x10
[lw:03495] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7f790956e390]
[lw:03495] [ 1] sst(_ZN3SST4Link15recvUntimedDataEv+0x11)[0x4a8d01]
[lw:03495] [ 2] /home/juddyoung/local/sstelements-8.0.0/lib/sst-elements-library/libSamba.so(_ZN3SST14SambaComponent5Samba4initEj+0x53)[0x7f78f97554f3]
[lw:03495] [ 3] sst(_ZN3SST10Simulation10initializeEv+0x64)[0x4b1124]
[lw:03495] [ 4] sst[0x45ce1a]
[lw:03495] [ 5] sst(main+0xee7)[0x450797]
[lw:03495] [ 6] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f790821e830]
[lw:03495] [ 7] sst(_start+0x29)[0x45c869]
[lw:03495] *** End of error message ***
Segmentation fault (core dumped)
this is gdb output:
juddyoung@lw:~/mzl$ gdb /home/juddyoung/local/sstcore-8.0.0/bin/sst
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/juddyoung/local/sstcore-8.0.0/bin/sst...done.
(gdb) r /home/juddyoung/mzl/bzip2.py 
Starting program: /home/juddyoung/local/sstcore-8.0.0/bin/sst /home/juddyoung/mzl/bzip2.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 3525 is executing new program: /home/juddyoung/local/sstcore-8.0.0/libexec/sstsim.x
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff2a71700 (LWP 3529)]
Creating Core 0 in Group 0
Creating Core 1 in Group 0
Creating Core 2 in Group 0
Creating Core 3 in Group 0
Creating Core 4 in Group 0
Creating Core 5 in Group 0
Creating Core 6 in Group 0
Creating Core 7 in Group 0
Initialized with 8 cores
Before initialization 
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
Assigning the PTW correctly
After initialization 

Thread 1 "sstsim.x" received signal SIGSEGV, Segmentation fault.
SST::Link::recvUntimedData (this=0x0) at link.cc:179
179	    if ( untimedQueue == NULL ) return NULL;

jjwilke, 2019-11-26:
During initialization (so the simulation hasn't run and completed yet), you're accessing a `nullptr` link.  `nullptr` links are legal in SST, but in the case of Samba we don't actually intended for any of the links to be null. What's happening is that you're specifying a particular core count with the `"corecount"` parameter, but then not creating all the necessary links in the python file.

Probably we should have some asserts or error checks in the Samba constructor to make sure the links are non-null. Same with the output, can you directly paste your Python file into the issue instead of posting attachments? It should be easy enough to pick out which links you're missing.

juddyoung, 2019-11-27:
@jjwilke

juddyoung, 2019-11-27:
@jjwilke

juddyoung, 2019-11-27:
@jjwilke 
import sst
import os

cores_per_group = 4*2 #2 cores and 2 MMUs
groups = 1
ring_latency = "50ps"

sst.setProgramOption("timebase", "1ps")

ariel_params={
        "verbose": "0",
        "maxcorequeue": "256",
        "maxissuepercycle": "2",
        "pipetimeout": "0",
        "corecount": groups * cores_per_group/2,
        "executable": "/home/juddyoung/spec/cpu2006/benchspec/CPU2006/401.bzip2/exe/bzip2_base.amd64-m64-gcc43-nn",
        "appargcount": 1,
        "apparg0": "/home/juddyoung/spec/cpu2006/benchspec/CPU2006/401.bzip2/data/ref/input/chicken.jpg",
        "arielmode": "1",
        "memmgr.memorylevels": "1",
        "memmgr.defaultlevel": "0"
}

ariel = sst.Component("a0", "ariel.ariel")
ariel.addParams(ariel_params)

l1_params ={
        "cache_frequency" : "2 Ghz",
        "cache_size" : "64 KB",
        "coherence_protocol" : "MSI",
        "replacement_policy" : "lru",
        "associativity" : "8",
        "access_latency_cycles" : "1",
        "cache_line_size" : "64",
        "L1" : "1",
        "debug" : "0"
}

#corecount = 4;

l1cache = sst.Component("l1cache", "memHierarchy.Cache")
l1cache.addParams(l1_params)

memory_params={
        "coherence_protocol": "MSI",
        "backend.access_time": "10ns",
        "backend.mem_size": "2048MiB",
        "clock": "1GHz",
        "use_dramsim": "0",
        "device_ini": "DDR3_micron_32M_8B_x4_sg125.ini",
        "system_ini": "system.ini"
}

memory = sst.Component("memory", "memHierarchy.MemController")
memory.addParams(memory_params)

mmu_params={
        "os_page_size": 4096,
        "corecount": groups * cores_per_group,
        "sizes_L1": 1,
        "page_size1_L1": 4,
        "assoc1_L1": 4,
        "size1_L1": 64,
        "sizes_L2": 1,
        "page_size1_L2": 4,
        "assoc1_L2": 8,
        "size1_L2": 512,
        "clock": "2 Ghz",
        "levels": 2,
[... 73 lines cut ...]
# Statistic defaults to Accumulator
ariel.enableStatistics([
      "cycles",
      "active_cycles",
      "instruction_count",
      "read_requests",
      "write_requests",
      "total_cores"
])

memory.enableStatistics([
      "total_cycles"
])

l1cache.enableStatistics([
      "CacheHits",
      "CacheMisses"
])

mmu.enableStatistics([
      "tlb_hits",
      "tlb_misses",
      "write_requests",
      "total_waiting"
])

jjwilke, 2019-11-27:
@gvoskuilen Not sure if this is based on a template you recognize

Why are you only creating links for half the cores? 
````
if next_core_id < cores_per_group * groups / 2:
````
You could also modify the Samba constructor and add asserts for `mmu_to_cache[i]` and `cpu_to_mmu[i]` to see which link is null.

juddyoung, 2019-11-28:
@jjwilke 
import sst
import os

next_core_id = 0
cores_per_group = 2*2 #2 cores and 2 MMUs
groups = 1
ring_latency = "100ps"

sst.setProgramOption("timebase", "1ps")

ariel_params={
        "verbose": "0",
        "maxcorequeue": "256",
        "maxissuepercycle": "2",
        "pipetimeout": "0",
        "corecount": groups * cores_per_group/2,
        "executable": "/home/juddyoung/spec/cpu2006/benchspec/CPU2006/401.bzip2/exe/bzip2_base.amd64-m64-gcc43-nn",
        "appargcount": 1,
        "apparg0": "/home/juddyoung/spec/cpu2006/benchspec/CPU2006/401.bzip2/data/ref/input/chicken.jpg",
        "arielmode": "1",
        "memmgr.memorylevels": "1",
        "memmgr.defaultlevel": "0"
}

ariel = sst.Component("a0", "ariel.ariel")
ariel.addParams(ariel_params)

l1_params ={
        "cache_frequency" : "2 Ghz",
        "cache_size" : "64 KB",
        "coherence_protocol" : "MSI",
        "replacement_policy" : "lru",
        "associativity" : "8",
        "access_latency_cycles" : "1",
        "cache_line_size" : "64",
        "L1" : "1",
        "debug" : "0"
}

#corecount = 4;

l1cache = sst.Component("l1cache", "memHierarchy.Cache")
l1cache.addParams(l1_params)

memory_params={
        "coherence_protocol": "MSI",
        "backend.access_time": "10ns",
        "backend.mem_size": "2048MiB",
        "clock": "1GHz",
        "use_dramsim": "0",
        "device_ini": "DDR3_micron_32M_8B_x4_sg125.ini",
        "system_ini": "system.ini"
}

memory = sst.Component("memory", "memHierarchy.MemController")
memory.addParams(memory_params)

mmu_params={
        "os_page_size": 4096,
        "corecount": groups * cores_per_group,
        "sizes_L1": 1,
        "page_size1_L1": 4,
        "assoc1_L1": 4,
        "size1_L1": 64,
        "sizes_L2": 1,
        "page_size1_L2": 4,
        "assoc1_L2": 8,
        "size1_L2": 512,
        "clock": "2 Ghz",
[... 269 lines cut ...]
in Samba.cc constructor, cpu_to_mmu 1 is null
in Samba.cc constructor, mmu_to_cache 1 is null
Assigning the PTW correctly
in Samba.cc constructor, cpu_to_mmu[2] is 0
in Samba.cc constructor, mmu_to_cache[2] is 0
Assigning the PTW correctly
in Samba.cc constructor, cpu_to_mmu[3] is 0
in Samba.cc constructor, mmu_to_cache[3] is 0
After initialization 
get into Samba.cc's init()
initializing core 0
[lw:32078] *** Process received signal ***
[lw:32078] Signal: Segmentation fault (11)
[lw:32078] Signal code: Address not mapped (1)
[lw:32078] Failing at address: 0x30
[lw:32078] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390)[0x7f384aeef390]
[lw:32078] [ 1] sst(_ZN3SST4Link15recvUntimedDataEv+0x1d)[0x4a8d0d]
[lw:32078] [ 2] /home/juddyoung/local/sstelements-8.0.0/lib/sst-elements-library/libSamba.so(_ZN3SST14SambaComponent5Samba4initEj+0x90)[0x7f383b0f05e0]
[lw:32078] [ 3] sst(_ZN3SST10Simulation10initializeEv+0x64)[0x4b1124]
[lw:32078] [ 4] sst[0x45ce1a]
[lw:32078] [ 5] sst(main+0xee7)[0x450797]
[lw:32078] [ 6] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f3849b9f830]
[lw:32078] [ 7] sst(_start+0x29)[0x45c869]
[lw:32078] *** End of error message ***
Segmentation fault (core dumped)