sstsimulator/sst-elements #1413: Error while running the multicore.py example included with ariel

kamilkhan52, 2020-01-10:
## 1 - Detailed description of problem or enhancement
I set up SSTv9.1 using the build instructions and given tarballs. The ariel examples `ariel_ivb.py`, `ariel_snb_mlm.py` and `ariel_snb.py run` to completion without errors. But when I try running the example multicore.py, it outputs an error related to ports:
`kamil@epic:(...) simple/examples$ sst multicore.py`
`Done configuring SST model`
`FATAL:  SST Core: ERROR:  Attempting to connect to unknown port: port0, in component membus of type memHierarchy.Bus.`
`FATAL:  SST Core: ERROR:  Attempting to connect to unknown port: port0, in component membus of type memHierarchy.Bus.`
`SST Fatal Backtrace Information:`
`    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x24c) [0x4ea59c]`
`    1 : sst(_ZN3SST11ConfigGraph24checkForStructuralErrorsEv+0x3f9) [0x4c4bc9]`
`    2 : sst(main+0x20f9) [0x498649]`
`    3 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f707e739b97]`
`    4 : sst(_start+0x2a) [0x4a527a]`
`MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 1.`

After reading through `sst-info memHierarchy.Bus`, it looks like the ports this component can use are defined to be:
`PORT 0 = low_network_%(low_network_ports)d (Ports connected to lower level caches (closer to main memory))`
`PORT 1 = high_network_%(high_network_ports)d (Ports connected to higher level caches (closer to CPU))`

However, I still could not get the code to work with some changes to port names.

## 2 - How to reproduce
The error can be reproduced by installing sst v9.x with ariel and running `sst-elements-library-9.1.0/src/sst/elements/ariel/frontend/simple/multicore.py`
## 3 - Operating system(s) and versions 
Kernel: 4.15.0-041500-generic x86_64 bits: 64 
compiler: gcc v: 7.2.0 
Desktop: MATE 1.22.0 
Distro: Linux Mint 19.2 Tina 
base: Ubuntu 18.04 bionic
### 4 - version of external libraries (Boost, MPI)
openmpi-2.1.3
boost-1.56

gvoskuilen, 2020-01-10:
@kamilkhan52 multicore.py hasn't been used in a long time and is no longer tested - I'll make a note to fix/remove it for the next release. The examples in ariel/frontend/simple/examples/stream/ should all work though.

kamilkhan52, 2020-01-10:
@gvoskuilen Thanks for the swift reply. 

I was more interested in multicore.py because I want to simulate a simple on-chip system with a Vaultsim memory backend. But it may be hard to get multicore.py to work, do you think I can get details about the system simulated by the other three ariel examples? Specifically about how the components are interfaced, and the changes I should make to for it to simulate a single socket on-chip system. Thanks!

gvoskuilen, 2020-01-10:
The ivb & snb configs are single socket systems. The configs refer to "groups" but that is just to construct the components in the same order that they're distributed on the on-chip ring network. You can ignore the 'inactive_cores' part of the configs. Here's a short description of each:

_ariel_ivb.py:_
12 cores, private per-core L1s and L2s, L1s & L2s have hardware prefetchers
Shared distributed L3 with 20 1MB slices, cachelines are interleaved (round-robin) across slices
Four memories, each with a directory in front of it.
Memory addresses are interleaved in 4KB chunks across the memories
Cores, L1s, and L2s are connected directly to each other
On chip network (ring) connects L2s, L3s, and directory/memories
The ring order is (core core core l3 l3 l3 l3 l3 mem) and repeat 4 times

_ariel_snb.py:_
Same as ivb except it has 8 cores and 4 memories
The ring order is (core core l3 l3 l3 l3 l3 mem) and repeat 4 times

_ariel_snb_mlm.py:_
This tests some of the multi-level memory support in Ariel but is otherwise identical to ariel_snb.py.
Ariel divides its address region into two groups with 0-2GB in one group and 2GB-4GB in the other group. It allocates pages in group 0 by default but intercepts mallocs and allocates them group 1 if the 'malloc.txt' config indicates it should. If you want to be able to do this, I can explain further.

kamilkhan52, 2020-01-11:
@gvoskuilen Thank you so much! That is very helpful.

closed 2020-01-11 by kamilkhan52