sstsimulator/sst-elements #1415: Motifs & dragonfly non-minimal routing

SalvatoreDiGirolamo, 2020-01-12:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
I'm trying to run a motif over a dragonfly with valiant routing. However, the firefly module seems to expect the header packet to always arrive first, and it generates the following error when this is not true:

```
sst: nicRecvCtx.cc:63: bool SST::Firefly::Nic::RecvMachine::Ctx::processStdPkt(SST::Firefly::FireflyNetworkEvent*): Assertion `m_streamMap.find(srcKey) != m_streamMap.end()' failed.
[salvo-tp:28388] *** Process received signal ***
[salvo-tp:28388] Signal: Aborted (6)
[salvo-tp:28388] Signal code:  (-6)
[salvo-tp:28388] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x12890)[0x7f3ba40f8890]
[salvo-tp:28388] [ 1] /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7)[0x7f3ba2e65e97]
[salvo-tp:28388] [ 2] /lib/x86_64-linux-gnu/libc.so.6(abort+0x141)[0x7f3ba2e67801]
[salvo-tp:28388] [ 3] /lib/x86_64-linux-gnu/libc.so.6(+0x3039a)[0x7f3ba2e5739a]
[salvo-tp:28388] [ 4] /lib/x86_64-linux-gnu/libc.so.6(+0x30412)[0x7f3ba2e57412]
[salvo-tp:28388] [ 5] /home/sst/test/sst-elements/build/lib/sst-elements-library/libfirefly.so(_ZN3SST7Firefly3Nic11RecvMachine3Ctx13processStdPktEPNS0_19FireflyNetworkEventE+0x519)[0x7f3b8edb2019]
[salvo-tp:28388] [ 6] /home/sst/test/sst-elements/build/lib/sst-elements-library/libfirefly.so(_ZNSt17_Function_handlerIFvvEZN3SST7Firefly3Nic11RecvMachine18processNetworkDataEvEUlvE_E9_M_invokeERKSt9_Any_data+0x94)[0x7f3b8eda58e4]
[salvo-tp:28388] [ 7] /home/sst/test/sst-elements/build/lib/sst-elements-library/libfirefly.so(_ZN3SST7Firefly3Nic15handleSelfEventEPNS_5EventE+0x3e)[0x7f3b8ed99c6e]
[salvo-tp:28388] [ 8] sst(_ZN3SST10Simulation3runEv+0x217)[0x55fa8b0648f7]
[salvo-tp:28388] [ 9] sst(+0xa22ed)[0x55fa8b00a2ed]
[salvo-tp:28388] [10] sst(main+0xec6)[0x55fa8affaa66]
[salvo-tp:28388] [11] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7)[0x7f3ba2e48b97]
[salvo-tp:28388] [12] sst(_start+0x2a)[0x55fa8b009bea]
[salvo-tp:28388] *** End of error message ***
```

Is it possible to run motifs on a dragonfly with non-minimal routing? 

2 - Describe how to reproduce

```
sst --model-options="--topo=dragonfly --shape=4:96:5:5 --cmdLine=\"Init\" --cmdLine=\"Bcast iterations=2 count=512 \" --cmdLine=\"Fini\"" emberLoad.py
```

3 - What Operating system(s) and versions 
Ubuntu 18.04

4 - What version of external libraries (Boost, MPI)
Boost: 1.65

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
sst-core: 65bdec55
sst-elements: bd24676fa

feldergast, 2020-01-21:
Ember requires in order delivery of packets within a message.  You will need to specify merlin.reorderlinkcontrol as the Nic module.  Reorderlinkcontrol will make sure the packets are delivered in the order they were sent.

You specify this by adding the following into the --model-options parameter:

--param=\"nic:module=merlin.reorderlinkcontrol\"

feldergast, 2020-01-27:
@SalvatoreDiGirolamo, were you able to get the motifs running correctly using reorderlinkcontrol?  If so, I will close the issue.

SalvatoreDiGirolamo, 2020-01-27:
@feldergast thanks, that is what I was looking for!

closed 2020-01-27 by SalvatoreDiGirolamo