sstsimulator/sst-elements #1416: Error in autogen.sh while setting up SST v9.1 with Balar

kamilkhan52, 2020-01-21:
## Detailed description of the issue
I am trying to set up Balar with SST v9.1. Following the Detailed Build and Installation Instructions, I am encountering an error when I reach the autogen.sh step. The error reads:
`configure:18323: error: possibly undefined macro: AC_MSG_ERROR`
`    If this token and others are legitimate, please use m4_pattern_allow. `
`    See the Autoconf documentation.`
`    autoreconf: /usr/bin/autoconf failed with exit status: 1`

Peeking into the script suggests that GPGPUSim may not be found. Which brings me to another issue with the instructions. The balar setup steps say that the balar and sst-gpgpusim repo should be cloned to a location inside `SST_ELEMENTS_HOME`, but the sub directories look like those inside `SST_ELEMENTS_ROOT`. If I follow the instructions word by word, balar is not detected in the first step of autogen.sh. The error I am reporting is for when I use `SST_ELEMENTS_ROOT` instead of `SST_ELEMENTS_HOME` to build balar in. Here is the complete output of autogen.sh:

    xx@xx:~/scratch91balar/src/sst-elements-library-9.1.0$ ./autogen.sh 

    Included Element Libraries via Makefile in src/sst/elements/

    cacheTracer
    cassini
    CramSim
    firefly
    GNA
    hermes
    kingsley
    merlin
    shogun
    simpleElementExample
    simpleSimulation
    thornhill

    Included Element Libraries via config.m4 files in src/sst/elements/

    ariel
    balar
    ember
    memHierarchy
    Messier
    miranda
    Opal
    prospero
    Samba
    scheduler
    VaultSimC
    zodiac

    Ignored Element Libraries in src/sst/elements/

    Generating configure files...
    Running /usr/bin/libtoolize...
    configure:18323: error: possibly undefined macro: AC_MSG_ERROR
          If this token and others are legitimate, please use m4_pattern_allow.
          See the Autoconf documentation.
    libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `config'.
    libtoolize: copying file `config/ltmain.sh'
    libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `config'.
    libtoolize: copying file `config/argz.m4'
    libtoolize: copying file `config/libtool.m4'
    libtoolize: copying file `config/ltdl.m4'
    libtoolize: copying file `config/ltoptions.m4'
    libtoolize: copying file `config/ltsugar.m4'
    libtoolize: copying file `config/ltversion.m4'
    libtoolize: copying file `config/lt~obsolete.m4'
    libtoolize: putting libltdl files in LT_CONFIG_LTDL_DIR, `src/libltdl'.
    libtoolize: copying file `src/libltdl/COPYING.LIB'
    libtoolize: copying file `src/libltdl/README'
    libtoolize: copying file `src/libltdl/argz_.h'
    libtoolize: copying file `src/libltdl/argz.c'
    libtoolize: copying file `src/libltdl/loaders/dld_link.c'
    libtoolize: copying file `src/libltdl/loaders/dlopen.c'
    libtoolize: copying file `src/libltdl/loaders/dyld.c'
    libtoolize: copying file `src/libltdl/loaders/load_add_on.c'
    libtoolize: copying file `src/libltdl/loaders/loadlibrary.c'
    libtoolize: copying file `src/libltdl/loaders/shl_load.c'
    libtoolize: copying file `src/libltdl/lt__dirent.c'
    libtoolize: copying file `src/libltdl/lt__strl.c'
    libtoolize: copying file `src/libltdl/libltdl/lt__alloc.h'
    libtoolize: copying file `src/libltdl/libltdl/lt__dirent.h'
    libtoolize: copying file `src/libltdl/libltdl/lt__glibc.h'
    libtoolize: copying file `src/libltdl/libltdl/lt__private.h'
    libtoolize: copying file `src/libltdl/libltdl/lt__strl.h'
    libtoolize: copying file `src/libltdl/libltdl/lt_dlloader.h'
    libtoolize: copying file `src/libltdl/libltdl/lt_error.h'
    libtoolize: copying file `src/libltdl/libltdl/lt_system.h'
    libtoolize: copying file `src/libltdl/libltdl/slist.h'
    libtoolize: copying file `src/libltdl/loaders/preopen.c'
    libtoolize: copying file `src/libltdl/lt__alloc.c'
    libtoolize: copying file `src/libltdl/lt_dlloader.c'
    libtoolize: copying file `src/libltdl/lt_error.c'
    libtoolize: copying file `src/libltdl/ltdl.c'
    libtoolize: copying file `src/libltdl/ltdl.h'
    libtoolize: copying file `src/libltdl/slist.c'
    libtoolize: creating file `src/libltdl/Makefile.am'
    configure:18323: error: possibly undefined macro: AC_MSG_ERROR
          If this token and others are legitimate, please use m4_pattern_allow.
          See the Autoconf documentation.
    autoreconf: /usr/bin/autoconf failed with exit status: 1

## Describe how to reproduce
1. Set up SST core according to the detailed instructions provided.
2. Skip to setting up balar without setting up SST elements
3. Replace all instances of `SST_ELEMENTS_HOME` with `SST_ELEMENTS_ROOT` in the instructions for balar.
4. The error is encountered at the autogen.sh stage.
 
## What Operating system(s) and versions 
Kernel: 4.15.0-041500-generic x86_64 bits: 64
compiler: gcc v: 7.2.0
Desktop: MATE 1.22.0
Distro: Linux Mint 19.2 Tina
base: Ubuntu 18.04 bionic

## Version of external libraries (Boost, MPI)
openmpi-2.1.3
boost-1.56

## Complete list of terminal commands used
// set to correct gcc version of 7.3 before starting
// Using: gcc 7.3 and g++ 4.8
// skipping openmpi installation, make sure it is installed and PATH set as follows.
export PATH=$MPIHOME/bin:$PATH  
export MPICC=mpicc   
export MPICXX=mpicxx

export LD_LIBRARY_PATH=$MPIHOME/lib:$LD_LIBRARY_PATH
export DYLD_LIBRARY_PATH=$MPIHOME/lib:$DYLD_LIBRARY_PATH
export MANPATH=$MPIHOME/share/man:$DYLD_LIBRARY_PATH

// make directories
mkdir scratch91balar/ scratch91balar/src/ local91balar/

// Get SST core and install
cd $HOME/scratch91balar/src
wget https://github.com/sstsimulator/sst-core/releases/download/v9.1.0_Final/sstcore-9.1.0.tar.gz
tar xfz sstcore-9.1.0.tar.gz
cd $HOME/scratch91balar/src/sstcore-9.1.0

export SST_CORE_ROOT=$HOME/scratch91balar/src/sstcore-9.1.0
export SST_CORE_HOME=$HOME/local91balar/sstcore-9.1.0
./configure --prefix=$SST_CORE_HOME
make all
make install

export PATH=$SST_CORE_HOME/bin:$PATH
which sst 
sst --version

// SST elements - get files
cd $HOME/scratch91balar/src
wget https://github.com/sstsimulator/sst-elements/releases/download/v9.1.0_Final/sstelements-9.1.0.tar.gz
mv sstelements-9.1.0.tar.gz sst-elements-library-9.1.0.tar.gz
tar xfz sst-elements-library-9.1.0.tar.gz
cd $HOME/scratch91balar/src/sst-elements-library-9.1.0
export SST_ELEMENTS_ROOT=$HOME/scratch91balar/src/sst-elements-library-9.1.0
export SST_ELEMENTS_HOME=$HOME/local91balar/sstelements-9.1.0

// clone balar
cd $SST_ELEMENTS_ROOT/src/sst/elements # Instructions use SST_ELEMENTS_HOME, but it sounds like a mistake
git clone https://github.com/sstsimulator/balar.git $SST_ELEMENTS_ROOT/src/sst/elements/balar
cd $SST_ELEMENTS_ROOT/src/sst/elements/balar
git checkout master

// clone sst-gpgpusim-externa
cd $SST_ELEMENTS_ROOT/src/sst/elements/balar
git clone https://github.com/purdue-aalp/sst-gpgpusim-external $SST_ELEMENTS_ROOT/src/sst/elements/balar/sst-gpgpusim
cd $SST_ELEMENTS_ROOT/src/sst/elements/balar/sst-gpgpusim
git checkout master

// check balar prereqs
// install cuda if required and update PATH and LD_LIBRARY_PATH
export PATH=/usr/local/cuda-9.1/bin:$PATH
export LD_LIBRARY_PATH=/usr/local/cuda-9.1/lib64:$LD_LIBRARY_PATH
export CUDA_INSTALL_PATH=/usr/local/cuda-9.1

sudo apt-get install build-essential xutils-dev bison zlib1g-dev flex libglu1-mesa-dev
sudo apt-get install doxygen graphviz
sudo apt-get install python-pmw python-ply python-numpy libpng12-dev python-matplotlib
sudo apt-get install libxi-dev libxmu-dev freeglut3-dev
sudo apt-get update
sudo apt update --fix-missing

// build balar
cd $SST_ELEMENTS_ROOT/src/sst/elements/balar/sst-gpgpusim
export GPGPUSIM_ROOT=`pwd`
source setup_environment release
make

// pin is already set up
cd $HOME/scratch91balar/src
export PIN_HOME=$HOME/local/packages/pin-2.14-71313-gcc.4.4.7-linux
export INTEL_PIN_DIRECTORY=$PIN_HOME

// Build SST elements
cd $HOME/scratch91balar/src/sst-elements-library-9.1.0
./autogen.sh

allevin, 2020-01-21:
@kamilkhan52, thank you for your detailed report.  I recently made some changes to the documentation and incorrectly identified $SST_ELEMENTS_HOME instead of $SST_ELEMENTS_ROOT.  

I believe that I have corrected the errors in the build instructions for SST-Core, SST-Elements, and balar.  Additionally, I added a few minor clarifying comments in these sections.

Also Note, while I reviewed your configure line for SST-Core is missing **--disable-mpi** and **--disable-mem-pools** this is noted in the balar build documentation.

Please review the changes, and let me know if there are additional changes necessary.  I will close this issue in a few days if I don't see a response.

hughes-c, 2020-01-21:
@allevin Thanks for fixing that. I was just looking at the documentation and it all looked correct.

hughes-c, 2020-01-21:
@kamilkhan52 No, the instructions in the detailed build instructions should work. In fact, you should be able to build the sst-gpgpusim branch anywhere, you just need to ensure that the --with-gpgpusim flag is set properly. Having balar in the source directory with the other elements is the only requirement since it must be registered with sst-core and that's done through autotools at the moment.

That being said, I'm not convinced your error is related to balar. You can clearly see that it's found when you run autotools but there's something wrong with a macro...and that could be anywhere. Can you post the few lines before and after 18323 in your configure file?

allevin, 2020-01-21:
@kamilkhan52 can you give us what OS version and version of autotools you have (autoconf --version)

kamilkhan52, 2020-01-21:
Hi @hughes-c 
Here is the section of code from my configure file:

      if test "$have_pin" = 1; then :
      sst_check_ariel="yes"
    else
      sst_check_ariel="no"
    fi
      if test "$sst_check_ariel" = "yes"; then :
      config_ariel=1
    else
      config_ariel=0
    fi

       if test $config_ariel -eq 1; then :
      active_element_libraries="$active_element_libraries ariel"
    fi
       ac_config_files="$ac_config_files src/sst/elements/ariel/Makefile"

       dist_element_libraries="$dist_element_libraries ariel"

      sst_check_balarx="yes"

      SST_CHECK_GPGPUSIM(have_gpgpusim=1,have_gpgpusim=0,AC_MSG_ERROR([GPGPU-Sim required, but not found]))

      if test "$have_gpgpusim" = 1; then :
      sst_check_balarx="yes"
    else
      sst_check_balarx="no"
    fi
      if test "$sst_check_balarx" = "yes"; then :
      config_balar=1
    else
      config_balar=0
    fi

Line 18323 is `SST_CHECK_GPGPUSIM(have_gpgpusim=1,have_gpgpusim=0,AC_MSG_ERROR([GPGPU-Sim required, but not found]))`

The following is from `src/sst/elements/balar/configure.m4` :

    dnl -*- Autoconf -*-

    AC_DEFUN([SST_balar_CONFIG], [
      sst_check_balarx="yes"

      SST_CHECK_GPGPUSIM([have_gpgpusim=1],[have_gpgpusim=0],[AC_MSG_ERROR([GPGPU-Sim required, but not found])])

      AS_IF([test "$have_gpgpusim" = 1], [sst_check_balarx="yes"], [sst_check_balarx="no"])
      AS_IF([test "$sst_check_balarx" = "yes"], [$1], [$2])])

@allevin I am using `autoconf (GNU Autoconf) 2.69`

kamilkhan52, 2020-01-21:
`GPGPUSIM_ROOT` is `/home/kamil/scratch91balar_revised/src/sst-elements-library-9.1.0/src/sst/elements/balar/sst-gpgpusim`

kamilkhan52, 2020-01-21:
Here is the OS and kernel versions:
Kernel: 4.15.0-041500-generic x86_64 bits: 64
compiler: gcc v: 7.2.0
Desktop: MATE 1.22.0
Distro: Linux Mint 19.2 Tina
base: Ubuntu 18.04 bionic

hughes-c, 2020-01-21:
@kamilkhan52 @allevin I can reproduce this with the tar file. I do not have a solution yet.

hughes-c, 2020-01-21:
Well, it looks like we shipped the tarball and then changed the way balar is built...and it now expects this macro to be defined: `SST_CHECK_GPGPUSIM`. The problem is that we added this definition, after 9.1, in sst_check_gpgpusim.m4.

Since you're building gpgpusim anyway, I would suggest skipping the tarballs and pulling everything from github. If you pull these two repos, you should be able to follow the rest of the build instruction -- setting the environment variables accordingly.
git clone https://github.com/sstsimulator/sst-core.git
git clone https://github.com/sstsimulator/sst-elements.git

hughes-c, 2020-01-21:
@allevin Even the balar commit tagged for release, https://github.com/sstsimulator/balar/commit/22ab0fc7dba15a29ac37e0e41b6c6a419459b152, uses that macro.

kamilkhan52, 2020-01-21:
I pulled the two repos from github instead of using the tarballs as you suggested. The error about the macro does not occur now but running configure prints "no" infront of gpgpusim and balar. 

    kamil@epic:~/scratch91balar_revised/src/sst-elements$ ./autogen.sh

    Included Element Libraries via Makefile in src/sst/elements/
    cacheTracer
    cassini
    CramSim
    firefly
    GNA
    hermes
    kingsley
    merlin
    pyproto
    shogun
    simpleElementExample
    simpleSimulation
    thornhill
    vanadis

    Included Element Libraries via config.m4 files in src/sst/elements/

    ariel
    balar
    ember
    memHierarchy
    Messier
    miranda
    Opal
    prospero
    Samba
    scheduler
    VaultSimC
    zodiac

    Ignored Element Libraries in src/sst/elements/
    Generating configure files...
    Running /usr/bin/libtoolize...
    libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, `config'.
    libtoolize: copying file `config/ltmain.sh'
    libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `config'.
    libtoolize: copying file `config/argz.m4'
    libtoolize: copying file `config/libtool.m4'
    (removed lines)
    ...
    ...

    libtoolize: copying file `src/libltdl/slist.c'
    libtoolize: creating file `src/libltdl/Makefile.am'
    kamil@epic:~/scratch91balar_revised/src/sst-elements$ ./configure --        prefix=$SST_ELEMENTS_HOME --with-sst-core=$SST_CORE_HOME --with-pin=$PIN_HOME --with-cuda=$CUDA_ROOT --with-gpgpusim=$GPGPUSIM_ROOT 
    checking for a BSD-compatible install... /usr/bin/install -c
    checking whether build environment is sane... yes
    checking for a thread-safe mkdir -p... /bin/mkdir -p
    checking for gawk... gawk
    checking whether make sets $(MAKE)... yes
    checking whether make supports nested variables... yes
    checking how to create a pax tar archive... gnutar
    checking whether make supports nested variables... (cached) yes
    checking for gcc... gcc
    checking whether the C compiler works... yes
    checking for C compiler default output file name... a.out
    checking for suffix of executables... 
    checking whether we are cross compiling... no
    checking for suffix of object files... o
    checking whether we are using the GNU C compiler... yes
    checking whether gcc accepts -g... yes
    checking for gcc option to accept ISO C89... none needed
    checking whether gcc understands -c and -o together... yes
    checking for style of include used by make... GNU
    checking dependency style of gcc... gcc3
[... 127 lines cut ...]
       if test "$sst_check_gpgpusim_happy" = "no"; then :
    
    else

          if test "$sst_check_cuda_happy" = "no"; then :
      as_fn_error $? "valid Cuda required for GPGPU-Sim" "$LINENO" 5
    fi
    
    fi

I continued with the rest of the procedure to build sst-elements and ran a balar test included in the repo. SST throws an error about the balar.balar component.

    Completed configuring the cuda-test model
    Opening element library balar failed: file not found
    Opening and resolving references for element library balar failed:
    	/home/kamil/local91balar_revised/sstelements-9.1.0/lib/sst-elements-library/libbalar.so: cannot open shared object file: No such file or directory
    Valid Components: 
    FATAL:  SST Core: can't find requested component or subcomponent 'balar.balar'
     SST Fatal Backtrace Information:
        0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x21c) [0x4d807c]
        1 : sst(_ZN3SST7Factory15isPortNameValidERKSsS2_+0x445)     [0x4cd6b5]
        2 : sst(_ZN3SST11ConfigGraph24checkForStructuralErrorsEv+0x337) [0x4b02d7]
        3 : sst(main+0x37f) [0x48a53f]
        4 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f263387eb97]
        5 : sst(_start+0x2a) [0x497cfa]

allevin, 2020-01-22:
@kamilkhan52 The "no's" in the configuration summary indicates that gpgpusim was not found, and therefore, balar will not be compiled.  This is why the balar.balar component is not found (it was never built).   Are you sure you configured with --with-gpgpusim=$GPGPUSIM_ROOT 

We have figured out why the sst_check_gpgpusim.m4 was not in the 9.1 tarball, and will be correcting the issue for SST 10.0 (spring release).  In the meantime, you will need to manually copy the sst_check_gpgpusim.m4. file to your sst_source_dir/config directory.  

We will be working on a documentation change for 9.1 to address this soon.

kamilkhan52, 2020-01-22:
@allevin Thanks for the explanation. Yes I did configure --with-gpgpusim=$GPGPUSIM_ROOT 

    kamil@epic:~/scratch91balar_revised/src/sst-elements$ ./configure --    prefix=$SST_ELEMENTS_HOME --with-sst-core=$SST_CORE_HOME --    with-pin=$PIN_HOME --with-cuda=$CUDA_ROOT --with-    gpgpusim=$GPGPUSIM_ROOT
    
This is how my environment variables looked:

    kamil@epic:~/scratch91balar_revised/src/sst-elements$ echo $GPGPUSIM_ROOT 
    /home/kamil/scratch91balar_revised/src/sst-elements/src/sst/elements/balar/sst-gpgpusim
    kamil@epic:~/scratch91balar_revised/src/sst-elements$ echo $PIN_HOME 
    /home/kamil/local/packages/pin-2.14-71313-gcc.4.4.7-linux
    kamil@epic:~/scratch91balar_revised/src/sst-elements$ echo $CUDA_ROOT 
    /usr/local/cuda-9.1

Thank you for the heads up about sst_check_gpgpusim.m4. I do see it in sst-elements/config now that I am cloning the repo instead of using the tarball.

allevin, 2020-01-22:
@kamilkhan52 The documentation for balar has been updated, I added a note explaining how to copy the sst_check_gpgpusim.m4 and also we changed instructions to checkout the release tag instead of master.  

I will discuss with @hughes-c on your build issue.

hughes-c, 2020-01-26:
@kamilkhan52 Can you grep your configure output and tell me what this says?
`checking for cudart_mod in`

I have a feeling that you either didn't compile sst-gpgpusim or didn't source the environment.

kamilkhan52, 2020-01-27:
Hi @hughes-c 
Thanks for pointing this out. It was looking in the wrong directory for `libcudart_mod.so`

    checking for cudart_mod in /home/kamil/scratch91balar_revised/src/sst-elements/src/sst/elements/balar/sst-gpgpusim/lib/gcc-7/cuda-9010/release... libcudart_mod.so not found

The correct directory is 

    /home/kamil/scratch91balar_revised/src/sst-elements/src/sst/elements/balar/sst-gpgpusim/lib/gcc-7.4.0/cuda-9010/release/

Where it has `gcc-7.4.0/` instead of `gcc-7/`

After creating a symlink between the two, balar is now detected. 

    checking for cudart_mod in /home/kamil/scratch91balar_revised/src/sst-elements/src/sst/elements/balar/sst-gpgpusim/lib/gcc-7/cuda-9010/release... yes

The example `cuda_test.py` does run into error without modification though. 

    ArielComponent[arielcpu.cc:564:forkPINChild] Launching executable: /home/kamil/local/packages/pin-2.14-71313-gcc.4.4.7-linux/intel64/bin/pinbin...
    SSTARIEL: Loading Ariel Tool to connect to SST on pipe: /sst_shmem_11720-0-1804289383 max instruction count: 1000000000 max core count: 1
    SSTARIEL: Function profiling is disabled.
    ARIEL-SST: Did not find ARIEL_OVERRIDE_POOL in the environment, no override applies.
    Failed to open IPC region '': Invalid argument
    FATAL: ArielComponent[arielcpu.cc:537:forkPINChild] Launching trace child failed!  Child Exited with status 1
    SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x21c) [0x4d807c]
    1 : /home/kamil/local91balar_revised/sstelements-9.1.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU12forkPINChildEPKcPPcRSt3mapISsSsSt4lessISsESaISt4pairIKSsSsEEE+0x56d) [0x7fea792f51dd]
    2 : /home/kamil/local91balar_revised/sstelements-9.1.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU4initEj+0xa9) [0x7fea792f5359]
    3 : sst(_ZN3SST10Simulation10initializeEv+0x64) [0x4df524]
    4 : sst() [0x498477]
    5 : sst(main+0x1196) [0x48b356]
    6 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7fea79b73b97]
    7 : sst(_start+0x2a) [0x497cfa]

kamilkhan52, 2020-01-27:
Update on the last comment:
I rebuilt SST-core and SST-elements and the issue has disappeared now. I did not add or remove code (just a print statement)

hughes-c, 2020-01-27:
@kamilkhan52 Thanks! 

@alevine947 Do you want to close this or leave it open until we find a solution to makedist?

allevin, 2020-01-28:
@kamilkhan52 @hughes-c I am closing this issue, as the specific problem has been resolved.  

I have created a new issue #1421 that will drive our longer term solution.

closed 2020-01-28 by allevin