sstsimulator/sst-elements #1419: Merlin: Dragonfly_legacy VS Dragonfly VS Dragonfly2

yao-kang, 2020-01-24:
Just would like to know if there are any documents discuss the difference between these three Dragonfly (Dragonfly_legacy, Dragonfly, Dragonfly2) implementations in Merlin

Thanks

yao-kang, 2020-01-25:
It seems that dragonfly_legacy is deprecated. Dragonfly and Dragonfly2 are the same.

feldergast, 2020-01-27:
That's correct.  It's a complex history.  The first dragonfly model was not very flexible, so we introduced dragonfly2.  It eventually didn't make sense to use the original dragonfly model, but to provide a short duration of backward compatibility, we renamed dragon to dragonfly_legacy and copied dragonfly2 to dragonfly.  As of SST 10, only dragonfly will remain.

closed 2020-01-27 by feldergast