sstsimulator/sst-elements #1421: Balar fails to properly build due to missing sst_check_gpgpusim.m4 in distribution

allevin, 2020-01-28:
1 - Detailed description of problem or enhancement
In the SST 9.1 sst-elements distribution tarbar; the sst_check_gpgpusim.m4 is not distributed causing the build of balar to fail because it cannot detect gpgpusim.  This problem does not exist in the devel or master branches of sst-elements.

The current workaround is now defined in the [SST detailed build instructions](http://sst-simulator.org/SSTPages/SSTBuildAndInstall9dot1dot0SeriesDetailedBuildInstructions/)

2 - Describe how to reproduce
Attempt to build balar from the tarbar.  Verify that the sst_check_gpgpusim.m4 is located in the sst-element/config directory.  

3 - What Operating system(s) and versions 
SST 9.1, all OS's that support balar

4 - What version of external libraries (Boost, MPI)
GPGPUSim as described in the SST Detailed build instructions related to balar [here](http://sst-simulator.org/SSTPages/SSTBuildAndInstall9dot1dot0SeriesDetailedBuildInstructions/)

When creating a release of SST, the make dist option of configure is run.  This will look at all elements in the tree and copy their required .m4 files into the tarball.  Since balar is not in the tree when running make dist, the sst_check_gpgpusim.m4 is not distributed.

This is actually an issue with balar being an standalone element but having dependencies on pieces inside of sst-elements.

I suggest that this can be fixed in multiple ways (from relatively easy up to difficult):
* `1` Force sst-elements to distribute all .m4 files in the sst-elements/config directory.  This is a bit of a shotgun approach, but it will resolve the issue for balar.
* `2` ship the  sst_check_gpgpusim.m4 as part of the balar repo, and have the user manually move the file into the sst-elements/config before running autogen.sh.  This will still require balar to be inserted into the sst-elements tree.
* `3` Fix balar to be truly standalone element with its own methods of discovering gpgpusim.

allevin, 2020-10-19:
#1449 Fixes this issue by distributing all config files (the shotgun approach).  It still does not resolve the underlying issue.

hughes-c, 2021-07-15:
@allevin Is this still an issue after merging balar into sst-elements?

allevin, 2021-07-15:
@hughes-c after merging balar, this would not be a problem anymore for balar.   The shotgun approach will also fix any other external element that requires to be placed in the sst-elements tree.   I don't anticipate that we will roll this change back.  

Ultimately, we need to have a standardized concept for external elements to be truly independent and perform their own discovery of dependancies.

I feel that we can close this issue.

closed 2021-07-15 by hughes-c